JMRI/JMRI #11880: resources\icons\markers does not include the local resources

KenC57, 2023-03-27:
User was adding a custom icon (CTC panel lockout) and found it only pulled from the system folder and not the local folder from the profile. The norm was to merge the contents from the local path icons to supplement the system icons.

DanielBoudreau, 2023-03-27:
The user hijacked the marker icon by modifying the gif.  Today the path to the marker icon isn't modifiable by a user.  One solution would allow a user to provide an alternate path to a marker icon.

I believe a better solution would allow the user to create panel objects that don't lock in place after the editor is closed.

bobjacobsen, 2023-03-27:
The catalog/NamedIcon class isn't looking things up in both locations. That surprises me, because I don't understand why the user-first lookup is working for other icons in that case.  Still investigating.

Pending what I find, I think I should fix this in any case.  

Agree that a more general solution would be good, but that'll take more time than I have right now.

DanielBoudreau, 2023-03-27:
Doing some quick testing using the Panel Editor, I found that I could use the option “Lock Position” to allow an icon to move when other items are locked.

So no issue when using Panel Editor, you can create a movable icon like a marker.

Heading out to ski, so I don’t have time to try the other editors.

Dan

closed 2023-03-29 by bobjacobsen