JMRI/JMRI #11914: Programming mode display ON programming panels

philguk, 2023-04-06:
Is it possible/practical/useful to somehow display the programming "mode" in operation in the window heading (where it now just says "Programming <roster ID>"??

I was thinking of "Programming on Main" or "Programming on Program Track" 

I often find myself in a programming session for a new loco and doubting whether I've selected "Programming Track" or "Main" (I typically ONLY use POM for updating existing locos and Program Track for setting up new ones from scratch - so my "last used" can oscillate between the two)

bobjacobsen, 2023-04-06:
See PR #11915

philguk, 2023-04-07:
Cool. Let me know if you need a beta test

Phil G

bobjacobsen, 2023-04-07:
@philguk There are development installers at https://builds.jmri.org/jenkins/job/testreleases/job/better-dp-labeling/31/  It would great if you could try one of them.

philguk, 2023-04-07:
Thanks Bob

I’m away for a few days but I’ll have a go at installing on my Pi when I get back

Phil G

bobjacobsen, 2023-04-08:
Closed by #11915 which will be in JMRI 5.3.5

closed 2023-04-08 by bobjacobsen

philguk, 2023-04-10:
Awesome Bob

Thanks for adding this so quickly. Just what I needed

Phil G