JMRI/JMRI #11939: Support for up to 69 functions

philguk, 2023-04-14:
Lenz LZV200 firmware now supports operation of up to 69 decoder functions

Although not every situation needs so many, the proliferation of sound decoders with multiple output channels (and additional features) can exhaust 28 or 32 functions that are currently supported by JMRI

Other decoders, besides Lenz, woul also benefit from support for more functions. In my case, I have ESU installations that would benefit from at least 32

DecoderPro (function labels, function maps etc) and JMRI throttles need to be enhanced to support more functions (please)

Thanks

icklesteve, 2023-04-14:
Hi @philguk 

Much infrastructure is already in place and in use by some connection types.
To operate, which connection type ( ie. ThrottleManager ) are you using?

Steve.

bobjacobsen, 2023-04-14:
Some background for who-ever self-assigns this:

jmri.jmrix.AbstractThrottle has an alternate constructor that lets you specify the maximum number of functions.  Once you do that, AbstractThrottle's operation is automatic.  That's in use in 

```
java/src/jmri/jmrix/can/cbus/CbusThrottle.java:36
java/src/jmri/jmrix/ecos/EcosDccThrottle.java:41
java/src/jmri/jmrix/sprog/SprogCSThrottle.java:27
java/src/jmri/jmrix/sprog/SprogThrottle.java:28
```

All of those set the maximum number of functions to 32, not 63+, so it would be good to check the upstream code for WiThrottle, etc, to see whether there are additional limits there.

To bring this to the Lenz-specific throttles, jmri.jmrix.lenz.XNetThrottle would have to be updated to use that constructor, then have the necessary methods implemented to send the necessary message(s) to the command station.  

Has anybody seen any documentation of what the Lenz command station expects to see to control these higher-numbered functions?

Lionel642, 2023-04-14:
Hello;

JMRI Throttle UI side, the function panel is assigning as many function
buttons as returned by the throttle, so that should be ok.

Regards;
Lionel

bobjacobsen, 2023-04-14:
I've looked into this a bit more:
 - The JMRI Throttle definitely supports up two F68 in its interface
 - From reading the JMRI code, it looks like the WiThrottle protocol will handle up to F68 (not sure the apps will do that, though)
 - Much of the structure is in place for XPressNet sending of the throttle commands.
 
So the primary problem is to find out what messages JMRI needs to send to the XPressNet interface.  The best way to find these is for somebody to get a copy of the relevant documentation.  

It might be possible to use JMRI's monitor to sniff the messages when a real Lenz throttle commands these functions, but it also might not.  I don't know if the Lenz interfaces will forward those to JMRI.  

@philguk Would you be willing to try that?  Open JMRI, open its Monitor for your Lenz connection, turn on "show raw data", when turn on and off some functions. Best would be to capture: F29, F36, F37, F44, F45, F52, F53, F60, F61 and F68. Thanks in advance!

philguk, 2023-04-14:
ExpressNet (Lenz)

Phil G

philguk, 2023-04-14:
[Lenz_XpressNet_v4.0 (German).pdf](https://github.com/JMRI/JMRI/files/11234717/Lenz_XpressNet_v4.0.German.pdf)

Google translated version of the Lenz XpressNet v4.0 manual

bobjacobsen, 2023-04-14:
Thanks.  I'll work on this over the next couple days.

philguk, 2023-04-14:
Some more digging - I may be asking for more than I immediately need

LokSound 5 decoders support up to F31 it seems (32 functions)

DecoderPro allows me to specify LABELS for functions up to and including F31, but the JMRI throttles only show up to F28 (as does WiThrottle)

SO it seems there is at least maybe a disconnect between DecoderPro maximum function support and other parts of JMRI

Given that I'll only be able to test up to F31 maybe support for the further function groups could be deferred until someone is in a position to actually test the support?

bobjacobsen, 2023-04-14:
There are development installers available here:

https://builds.jmri.org/jenkins/job/testreleases/job/lenz_f68/52/

Would appreciate it if you could try them and see what works and what doesn't.  

Note:  This isn't final, there's some cleanup yet to do; think of these as a test of how well I can read the manual.

philguk, 2023-04-14:
> The limit on functions on a throttle is due to the layout hardware,
> i.e. /command station/.
>
> The limit on functions listed in DecoderPro is due to the capabilities
> of the /decoder/.
>
> So they're not really related to each other.
>
> It's not much harder to expand the Lenz support to F68 than to F32, so
> I'm working on doing that. Should have installers for you to test in
> the next day or two.

Gotcha

BUT I have a decoder that goes beyond F28 and a command station that 
does too, but the JMRI throttles stop at F28 :(

Phil G

philguk, 2023-04-14:
I'll try and take a look over the weekend. I have some other oddities 
going on that I'd like to bottom out first

Thanks for this

Phil G

crosland, 2023-04-14:
> From "philguk" ***@***.***>
> To "JMRI/JMRI" ***@***.***>
> Cc "Subscribed" ***@***.***>
> Date 14/04/2023 19:28:33
> Subject Re: [JMRI/JMRI] Support for up to 69 functions (Issue #11939)

Something at the back of my mind: Does it make a difference if you 
create a roster entry for the 32-Fn decoder, open that and then open a 
throttle?

Andrew

bobjacobsen, 2023-04-14:
> Does it make a difference if you create a roster entry for the 32-Fn decoder, open that and then open a throttle?

Empirically, no.  I just tried a couple cases of this and the number of functions was always taken from the system limit, not the roster entry.  So you can have 68 functions visible on that 32-function decoder.

philguk, 2023-04-14:
I assume the .tgz is the Linux one (for a RasPi)??

Phil G

bobjacobsen, 2023-04-14:
> I assume the .tgz is the Linux one (for a RasPi)??

Right.  If should expand into a “JMRI” directory that works the same as your existing one.  If you expand it in a different place than your current JMRI program directory, you can have both on the machine at the same time.

philguk, 2023-04-14:
No worries

I always rename my current /JMRI before I do any installs so fallback is 
just another rename :)

danielb987, 2023-04-14:
@bobjacobsen 
> The limit on functions on a throttle is due to the layout hardware, i.e. _command station_.

Yes and no. Yes about hardware, but I think it's a No about command station. I'm not entirely sure, but I think LocoNet uses a DCC command on the LocoNet bus for the high functions so for LocoNet I think 68 functions are available for all LocoNet command stations that can send the DCC command but maybe JMRI isn't yet sending this command for >28 functions. So it would be good to add this to LocoNet as well.

dheap, 2023-04-14:
Take a look at the work already done in:
#8932
#8933

Dave

bobjacobsen, 2023-04-14:
@dheap Thank you!  `maxFnNum` is just the right thing for controlling the number of functions displayed by the throttle. The Lenz connection here doesn't and in PR #11943 doesn't use raw packets, but those will be helpful for e.g. Digitrax and NCE.  Thanks again!

bobjacobsen, 2023-04-15:
@philguk (or anybody else with Lenz hardware) there are new development installers at

https://builds.jmri.org/jenkins/job/testreleases/job/lenz_f68/56/

Thanks in advance for testing these.

bobjacobsen, 2023-04-15:
The https://builds.jmri.org/jenkins/job/testreleases/job/lenz_f68/56/ installers include use of maxFnNum so that a JMRI throttle using a roster entry will display the lowest of maxFnNum or what the system supports.

philguk, 2023-04-15:
Cool!

I can confirm that my 32 function (LokSound5) decoder now has JMRI 
throttle support for all 32 functions, and displays the labels for all 
32 as well. And it operates all the functions correctly too

I'm using this latest(?) download that seems to be JMRI 5.3.5+Ra9cdcf927

WiTrottle still maxes out at F28, but that's probably a WiThrottle 
limitation rather than a JMRI issue - I'll mention it to Brett

Much appreciate the speedy response to this request. You kind of took me 
by surprise tbh

Oh, and thanks for the "you're programming on the main" splash at the 
bottom of DecoderPro. That's been invaluable when tweaking function maps!

Phil G

philguk, 2023-04-15:
Just to add

This is all tested on a RasPi (Steve Todd build, with updates)

Everything I have tried so far has worked (operating functions, driving 
with JMRI and WiThrottle, operating switches and routes and rudimentary 
LogixNG)

Phil G

bobjacobsen, 2023-04-15:
@philguk Thanks for the testing!  This'll get merged and be in the next test release.

bobjacobsen, 2023-04-15:
Closed by PR #11943

closed 2023-04-15 by bobjacobsen

philguk, 2023-04-15:
I can see I am going to have to be careful what I ask for in future!!

Thanks a ton Bob (again)

Enjoy the rest of your weekend

Phil G