JMRI/JMRI #11954: Create backup of tables and panels on storage

philguk, 2023-04-17:
When storing tables and panels to an existing table/panel file, the original is merely overwritten - and no backup is created of the old version

In a similar way to other files (rosters etc), it could be useful that a table/panel store could create a .bak version rather than simply overwriting the old file

This would also simply provide a 1 generation backup in case problems are encountered

philguk, 2023-04-17:
> Wrong. When storing PanelPro tables and panels, the existing file is
> moved to the backupPanels directory with the data and time added to
> the file name. Then a new file is created.
>
> Dave Sand

Ahh

Where IS this "backupPanels directory"???

I'm not finding anything on my RasPi

Phil G

dsand47, 2023-04-17:
> Ahh Where IS this "backupPanels directory"??? I'm not finding anything on my RasPi Phil G
> […](#)

With PanelPro running, go to "Help => File Locations".  Click on the "Open Profile Location" button.  That should open the default file manager at the correct location.

philguk, 2023-04-17:
This is actually already provided - I just didn't know (and didn't know where the panel file backups are actually created)

They are in the Backup Panels folder in the current profile folder

Issue closed

philguk, 2023-04-17:
See my final comment (above)

Already provided

philguk, 2023-04-17:
Cool!

Must say though, I would NEVER have looked there for a panel backup!!

I'll close this issue as "not needed"

Phil G

closed 2023-04-17 by philguk