JMRI/JMRI #11956: Tables/Lights Adding a Sequential Range Fails

Alan-Lomax, 2023-04-18:
Adding a single light works as expected. Adding a sequential range does not.
Panel Pro 5.1.6

ps:
The same functionality in 'sensors' works as expected both ways.

icklesteve, 2023-04-18:
Hi @Alan-Lomax 

I'm unable to reproduce this with Internal or CBUS Lights in HOM.

Which system connection type are you using?
What starting System Name are you starting from?

Ta, Steve.

Alan-Lomax, 2023-04-18:
Hi Steve,

Thanks for getting back to me.
Mine is a CMRI direct serial connection (to a RS-485 / USB adapter)
At the JMRI end I have configured CMRI/SMINI nodes.  At the remote ends 
it is Arduino Nano's using the CMRI.h library.
For now just two Arduino are set up on address 0 and address 1 for 
development and test.

When I configure the Light addresses one at a time it all works as expected.
I have built and tested addresses 1,2,3,4,5,6,7   and 1001-1007 
(Corresponding to node address 0 and node address 1 respectively)
Toggling the state in JMRI flips the corresponding bit on the Arduino.

When I tried adding a range - it failed every time.
The system names (auto defined) are CL1-CL7 and CL1001-CL1007 as shown 
below.

Alan-Lomax, 2023-04-21:
OK I understand what it is now. The clues were as you pointed out in the system console if I had known to look there.

The reported message was:
...  ERROR - inconsistent user '8' and system name 'CL9' results; user name related to CL8 [AWT-EventQueue-0]

Which led me directly to the fact that adding a range auto increments the system name but does nothing with the User Name.
Because I did in fact have a User Name specified this fails every time. The user name is enforced to be unique but is not incremented during a range build.
 
I suggest as an improvement:
If the user name is not blank then  append a -1 , -2, -3 during a range build to at least stop the operation from failing without any user feedback.

I will closed this off  with thanks for your help..

closed 2023-04-21 by Alan-Lomax