JMRI/JMRI #11963: Digitrax_0SFX_Series7 Decoder Definitions

SPJMaker, 2023-04-20:
[Digitrax_0SFX_Series7.zip](https://github.com/JMRI/JMRI/files/11287466/Digitrax_0SFX_Series7.zip)
Here is the draft new decoder definitions for Digitrax Series7 SDXH167 decoders.
CV10 should be in Motor Pane. I only did the sound related pages. So CV10 is in the Sound Pane

John McMasters
AnPRR Groups.io
864-933.7988
SPJmaker@gmail.com

bobjacobsen, 2023-04-20:
Just tried it, and "EMF Dynamic Config" is in use.  Sorry for the confusion.  But "EMF Option 1" works.

bobjacobsen, 2023-04-20:
in PR #11964.  I've attached the PDF description file here for future reference.

[Digitrax Series7 with JMRI DecoderPro.pdf](https://github.com/JMRI/JMRI/files/11288024/Digitrax.Series7.with.JMRI.DecoderPro.pdf)

bobjacobsen, 2023-04-20:
There are a number of places where the same item name is reused.  For example, "Sound Setting 16" occurs multiple times.  Can these be renumbered? "Sound Setting 1" through "Sound Setting 40" as are three groups of 16 more "Sound Group 1 Option 1" through option 16, groups 1 to 3.

bobjacobsen, 2023-04-20:
Here are the changes I need to make so the file can validate and to move CV=10 to the motor pane:
```
675c675
<     <variable item="Sound Project Code Version" CV="152" default="70" readOnly="yes" include="SDXH167,SDXH167Ext,SDXH167ExtS,SDXH167ExtD">
---
>     <variable item="Sound Project Code Version" CV="152" default="70" readOnly="yes" include="SDXH167,SDXH167Ext,SDXH167ExtS,SDXH167ExtD">>
688c688
<         <decVal max="100"/>
---
>         <decVal max="100"/>  speed at which brake sounds begin to play
741c741
< 	   <variable CV="10" item="EMF Option 1" default="0"   include="SDXH167,SDXH167Ext,SDXH167ExtS,SDXH167ExtD">
---
> 	   <variable CV="10" item="Sound Option 8" default="0"   include="SDXH167,SDXH167Ext,SDXH167ExtS,SDXH167ExtD">
766a767,772
> 	      <variable CV="148" item="Sound Setting 12" default="30"  exclude="SDXH167ExtS,SDXH167ExtD">
>         <decVal/>
>         <label>Compressor Start Rate</label>
>         <label xml:lang="it">Frequenza compressore Run/Popoff</label>
>         <tooltip>Time ms. between Compressor cycles</tooltip>
>       </variable>
```

SPJMaker, 2023-04-20:
Thanks Bob.
I hope everyone gets a lot of benefit from the new decoder and definition.

John McMasters
AnPRR
***@***.***

bobjacobsen, 2023-04-22:
Content merged by #11964 and will be in next test release in a couple weeks.

closed 2023-04-22 by bobjacobsen

SPJMaker, 2023-07-01:
Content merged by #11964 <https://github.com/JMRI/JMRI/pull/11964>  this
definition never got to the next test release?

John McMasters
AnPRR
***@***.***

bobjacobsen, 2023-07-01:
> Content merged by #11964 <https://github.com/JMRI/JMRI/pull/11964> this 
> definition never got to the next test release? 

It should be in [JMRI 5.3.6](https://www.jmri.org/releasenotes/jmri5.3.6.shtml) and later.  Do you not see it?

SPJMaker, 2023-07-01:
Got it. I had not cleaned out my local rr decoders folder.
I did a decoder index rebuild.
All is good.

There will be several new decoders in the Series7 family coming soon.

John McMasters
AnPRR
***@***.***