JMRI/JMRI #11969: Sript for AnyRail export PamelPro file and Additional information for the AnyRail page

Cliff-Anderson, 2023-04-21:
This is an outgrowth of the discussion at https://groups.io/g/jmriusers/topic/anyrailbuildsensorlist_script/98099598 and the suggestions from Bob Jacobsen and Peter Ulvestad.

The referenced script upload. The AnyRailBuildBlockSensorList.py script us attached here:
[AnyRailBuildBlockSensorList.zip](https://github.com/JMRI/JMRI/files/11299507/AnyRailBuildBlockSensorList.zip)

Suggested additional information for the https://www.jmri.org/community/connections/AnyRail/index.shtml pages is attached here: 
[AnyRailBuildBlockSensorList Description.docx](https://github.com/JMRI/JMRI/files/11299522/AnyRailBuildBlockSensorList.Description.docx)

My webpage skills are extremely limited and the Word document is embarrassing too long.
 
Additional posts to the Groups.io thread by Lawrence Varady suggest that this issue might become mute sometime in the future?

bobjacobsen, 2023-04-21:
Thank you! 

Now in PR #11966 and JMRI/website#509.  I took the .docx Word file as a PDF to capture the images and formatting.

danielb987, 2023-04-22:
> Now in PR #11966 and [JMRI/website#509](https://github.com/JMRI/website/pull/509). I took the .docx Word file as a PDF to capture the images and formatting.

A suggestion is to include the .docx file too in the website PR together with a note in the README file that this file is the source of the PDF. Preferable as an Open Document Format (.odt extension). I think it's easier for most people to edit the Word document than edit the PDF document. And .odt files can be edited with OpenOffice and LibreOffice as well.

bobjacobsen, 2023-04-22:
Done, thanks for the suggestion.

closed 2023-04-22 by bobjacobsen