JMRI/JMRI #11982: Soundtraxx tsunami-1 not recognized during "read type from decoder"

playswithcode, 2023-04-28:
Original tsunami decoder (cv7=64) not recognized by version 5.3.4, decoder was recognized by an earlier version of JMRI.

Operations used to encounter problem:
- NCE PowerPro v.165B with USB interface for programming track
- installed on Linux, using version 5.3.4+Raae2040ea3 with Java 11.0.18
- place locomotive on programming track, select "New Loco", then "read type from decoder"

Result - some CV's read, correctly identifies mfg 114, model 64, but cannot select decoder type.

Excerpt from messages.log is:
2023-04-25 14:44:54,331 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2023-04-25 14:44:54,342 util.Log4JUtil                        INFO  - This log is appended to file: /home/pn/.jmri/log/messages.log [main]
2023-04-25 14:44:54,343 util.Log4JUtil                        INFO  - This log is stored in file: /home/pn/.jmri/log/session.log [main]
2023-04-25 14:44:54,402 apps.AppsBase                         INFO  - DecoderPro version 5.3.4+Raae2040ea3 starts under Java 11.0.18 on Linux amd64 v5.10.0-21-amd64 at Tue Apr 25 14:44:54 EDT 2023 [main]
2023-04-25 14:44:59,211 gui3.Apps3                            INFO  - Starting with profile ProvNorthernMRR.3f7be4bc [main]
2023-04-25 14:44:59,311 node.NodeIdentity                     INFO  - Using 4539751f-7e9e-4829-9f45-c8daaf7f39c6 as the JMRI storage identity for profile id 3f7be4bc [AWT-EventQueue-0]
2023-04-25 14:44:59,424 xml.AbstractSerialConnectionConfigXml INFO  - Starting to connect for "NCE" [main]
2023-04-25 14:44:59,578 usbdriver.UsbDriverAdapter            INFO  - NCE USB ttyUSB0  port opened at 9600 baud [main]
2023-04-25 14:44:59,711 nce.NceConnectionStatus               INFO  - NCE EPROM revision = 7.3.0 [AWT-EventQueue-0]
2023-04-25 14:44:59,852 nce.NceConnectionStatus               INFO  - Connected to NCE command station [AWT-EventQueue-0]
2023-04-25 14:45:00,181 util.FileUtilSupport                  INFO  - File path program: is /home/pn/JMRI/ [main]
2023-04-25 14:45:00,182 util.FileUtilSupport                  INFO  - File path preference: is /home/pn/.jmri/ProvNorthernMRR.jmri/ [main]
2023-04-25 14:45:00,182 util.FileUtilSupport                  INFO  - File path profile: is /home/pn/.jmri/ProvNorthernMRR.jmri/ [main]
2023-04-25 14:45:00,182 util.FileUtilSupport                  INFO  - File path settings: is /home/pn/.jmri/ [main]
2023-04-25 14:45:00,182 util.FileUtilSupport                  INFO  - File path home: is /home/pn/ [main]
2023-04-25 14:45:00,183 util.FileUtilSupport                  INFO  - File path scripts: is /home/pn/JMRI/jython/ [main]
2023-04-25 14:57:28,087 symbolicprog.CombinedLocoSelTreePane  WARN  - Found mfg 141 (Throttle-Up (SoundTraxx)) version 64; no such decoder defined [AWT-EventQueue-0]
2023-04-25 14:57:28,090 decoderdefn.IdentifyDecoder           INFO  - Decoder returns mfgID:141;modelID:64;productID:-1 [AWT-EventQueue-0]
2023-04-25 15:05:36,146 jmrit.XmlFile                         INFO  - No /home/pn/.jmri/ProvNorthernMRR.jmri/roster/PN_1802.xml file to backup [AWT-EventQueue-0]

playswithcode, 2023-04-28:
Note, also posted in groups.io - one other user reports encountering the same issue.

See https://groups.io/g/jmriusers/message/216701

playswithcode, 2023-04-29:
Additional information - reset-to-factory on Tsunami decoder, and attempted operation as above.

Same results, decoder CV values of significance are:
CV8 = 114
CV7 = 64
CV105 = 11  (firmware version 1.1)
CV106 = 0

Note that CV105 and 106 are user-writable, thus the above default values might be changed by the user.

bobjacobsen, 2023-05-23:
A few questions:

1) What type of decoder is this?  I.e. what do you select that you think is the correct decoder definition?

2) Are you sure this worked in the past?  Which JMRI version?  I.e. how long ago?

Thanks

bobjacobsen, 2023-05-23:
Also, what are CV253 and CV256?  Thanks

bobjacobsen, 2023-05-23:
I think I’ve found it!  Whether or not the problem occurs depends on whether you have local copies of the decoderIndex or not, which was why I couldn’t see it.

Development installers are available here:

https://builds.jmri.org/jenkins/job/testreleases/job/hiding-tsunami/62/

I’d very much appreciate it if people who are seeing the problem could check whether one of these fixes it.

Thanks in advance.

Bob
—
Bob Jacobsen
***@***.***

connelly38, 2023-05-23:
That works Bob, thanks!

bobjacobsen, 2023-05-24:
It would be helpful to get one or two more tests done.  I don't want you to go to a bunch of trouble, hopefully a couple people will test and comment before Friday.  But if not, I'd appreciate the test.

No special logging needed. Just check to make sure that your decoders are now properly identifying.

If you do find a problem, I'd appreciate the CV7 and CV8 values, and which type you think the decoder is.

playswithcode, 2023-05-24:
I'll pull the linux tgz now, and get it loaded on my USB stick in preparation for going to the club on Friday and
collecting the data.

In order to properly identify a Tsunami-1, I have to pull the shell and use the Mk1 eyeball for the markings
in the silkscreen on the board. I'm reluctant to do that, since these are club-owned locomotives and I've
had trouble getting shells off of some engines in the past. (Hey, pass me that crowbar, wouldja?)

The following may be stating the obvious, but back when these were manufactured, it seems as though none of
the manufacturers understood the utility of being able to read CVs to get the specific productID - so they did
the minimum required CV8/CV7 and left it at that. Soundtraxx docs state that CV105 & 106 will contain information
about the firmware version, but they allow customers to change the values - so in my opinion, the contents cannot be
relied upon except immediately after a factory reset.

Newer versions of the Soundtraxx decoders (Tsunami2 & Econami) now use the manufacturer-specific CVs 253-256
to encode expanded information to identify the decoder.

bobjacobsen, 2023-05-25:
@playswithcode Thanks!  It looks like enough people have tested it at this point that I'm pretty sure it's working.

Agree that early decoders really couldn't be specifically identified.  They were simpler, so that was a bit less of a problem. DecoderPro can't really narrow down a first-generation Tsunami because of that, so it will show a lot of choices. DecoderPro can usually narrow down a newer Tsunami to just one or two.

bobjacobsen, 2023-05-25:
Fixed by PR #12051