JMRI/JMRI #11991: DCC-EX Fast Clock Interface

Asbelos, 2023-05-01:
DCC-EX has fast-clock automation facilities and can :
 a) react to an external clock (which could be JMRI) 
 b) broadcast the external clock time changes to attacked throttles (which includes JMRI) 
 
 To achieve this the DCC-EX interface in JMRI would need to handle the following commands/responses.
 
**From JMRI (or other external clock)  to DCC-EX:**
    `<JC m ss>`   
Sets DCC-EX clock to  m minutes after midnight and running at speed s  (e.g. s=4 means 15 seconds per minute). 
 If JMRI is acting as the master clock, this would be expected to be sent at the start of every fast-minute. 
 This will also cause DCC-EX to broadcast the time (see below) 
    
**From a throttle (or JMRI that is not acting as the master clock) to DCC-EX**
   `<JC>`
requests immediate broadcast response of current DCC-EX clock.(see below). This doesn't need to be polled, but at startup it helps to get a time while you wait for the next broadcast.
    
**Broadcast From DCC-EX to JMRI (or other throttles)**
     `<jC m s>` 
DCC-EX has been told by the external clock (or JMRI) that the time is now m minues past midnight and speed is s.

Asbelos, 2023-05-01:
Unfortunately DCCEX has no floating point runtime support due to memory and performance constraints. 😕 
Any suggestions?

danielb987, 2023-05-01:
> Unfortunately DCCEX has no floating point runtime support due to memory and performance constraints. confused
> Any suggestions?

A simple solution is to round the speed to the nearest integer. It's hopefully good enough.

mstevetodd, 2023-05-19:
My findings after a bit of code fiddling:
JMRI has a mechanism to choose a "Time Source". I have a "Time Source" dropdown with choices "Internal Computer Clock" and "DCCpp Fast Clock".
So we need 2 different behaviours here.
1) if TimeSource=Internal, then JMRI sends the <JC mmmm rr> at startup and for every fast minute change, and ignores all incoming <jC mmmm rr> messages.
2) if TimeSource=DCCpp, JMRI does not send any <JC mmmm rr> messages and updates the JMRI Fast Clock time and rate from all  incoming <jC mmmmm rr> messages. This option requires that the user has set up additional hardware/software to provide the FastClock functionality. It is not part of EX-CommandStation.

mstevetodd, 2023-05-19:
> > Unfortunately DCCEX has no floating point runtime support due to memory and performance constraints. confused
> > Any suggestions?
> 
> A simple solution is to round the speed to the nearest integer. It's hopefully good enough.

> Could the speed be a floating point number? I'm pretty sure JMRI supports fractional speeds, like 4.5.

There is a `requiresIntegerRate()` edit that will prevent UI entry of a fractional number.

mstevetodd, 2023-05-24:
I verified that FastClock times sent to DCC-EX are properly broadcast out to EngineDriver clients using the `<>` protocol, and properly to WiThrottle clients up to a point where we're getting an overflow in the PFT message. I've asked the DCC-EX folks to fix.

mstevetodd, 2023-05-24:
> 
> Should we hold #12053 until this is fixed or do you want the PR to be merged anyway?

No, please merge.

danielb987, 2023-05-24:
> > Should we hold #12053 until this is fixed or do you want the PR to be merged anyway?
> 
> No, please merge.

Will do once 24 hours has passed.

mstevetodd, 2023-05-25:
Closed by #12053

closed 2023-05-25 by mstevetodd