JMRI/JMRI #12008: Programmer error on read indexed CV at Programming on Main (POM)

uli-2016, 2023-05-08:
I use Decoder Pro with z21 at Programming on Main (POM). CV1 to 1021 are read without problems. When reading index CVs shows the following error message: "Programmer error: Requested operation not implemented in command station (303)". The CV for the index e.g. CV1021 was written correctly during this attempt, only the subsequent reading of the CV fails.
However, writing the index CV works for 80 percent of CVs, but with no discernible system. The ops delay is set 1000.

danielb987, 2023-05-10:
@uli-2016 
Have reading indexed CVs at POM worked before with the z21?

uli-2016, 2023-05-12:
"Have reading indexed CVs at POM worked before with the z21?" I don't know. Probably not.

The following sequence should run at CV read at POM (e.g. indexed CV257.31=8.32=3):
Write CV31=8
Read CV31, if not equal = 8 then ERROR
Write CV32=3
Read CV32, if not equal = 3 then ERROR
Read CV257
similar sequence on write an indexed CV

On Programming Track and with the Simple Programmer this sequence runs, but should also run with the standard Programming On Main.

danielb987, 2023-05-12:
@bobjacobsen 
Do you know anything about this? Or is this a question for the JMRI developers list?

bobjacobsen, 2023-05-12:
I don't understand the issue, but I think it would be better on jmri-developers where more experience can be brought to bear.

uli-2016, 2023-05-12:
I don't know if it's a problem with the program or my use of the program and my hardware.
I just don't understand that with POM programming reading and writing normal CVs works and with indexed CVs error messages come.

I still found out that the index CV was still written, but reading the indexed CV itself then fails.

dheap, 2023-05-12:
Program on Main shouldn't try to Read Back indexed CVs changes  to verify.

Dave

dheap, 2023-05-12:
> Program on Main shouldn't try to Read Back indexed CVs changes to verify.
> 
> Dave

Unless the Z21 is capable of POM CV Read (RailCom?),

Pugwash1, 2023-05-13:
The z21 can read back from the main.
The issue is it cannot write to the main very fast.
When writing cv31 and cv32  then read cv nn it needs to pause for about 100msec. Between each step . Turning the cv cache on in the roster minimizes the impact

uli-2016, 2023-05-14:
The CV cache is active on at my Decoder Pro. Could be help, to have a delay between write index CVs and read CV nn?

bobjacobsen, 2023-05-14:
Do we know that a delay would help this?  How long should it be?

When DP is writing a bunch of CVs, e.g. a write page operation, do they all get properly written?  Or does there need to be a short delay after every write CV?

uli-2016, 2023-05-14:
I made just now some test with several decoders.
Read an write of multiple CVs with "Read/Write full sheet" works with simple CVs. If there are problems you can increase the "Ops Mode Delayed Programming" in the decoder file. But this does nothing at all with induced variables".

Note: The option "When possible, confirm CV writes by following them with a read" should not be active at POM programming. This creates errors when write CVs

bobjacobsen, 2023-05-14:
It's possible to add a short delay after _every_ POM CV write.  Or a short delay before _every_ POM CV read.

The z21 programmer code in JMRI doesn't really know whether it's writing for an indexed variable or for a regular CV write.  It would be complex and error-prone to change that. 

Similarly, the code that handles indexed operations doesn't know what kind of programmer it's using.  That would also be complex and error-prone to change.

I'd be happy to code up a short delay after every POM write, or alternately a short delay before every POM read.  What should the delay time be?

uli-2016, 2023-05-14:
I am not sure about the right delay, I suggest to try 600 ms. I think a good solution would be to put these delay time as an option on the Preferences / Connection

[image]

bobjacobsen, 2023-05-14:
600 ms after every write sounds very long.  It would make things like write all sheets very much slower. I hope that big a delay isn’t necessary.

There are limits to how much work I want to do on this until we know for sure that timing is the issue.  So I’ll code something up where a script can change the time to allow you to experiment with it.  Might take a day or two to get it done.  Can you build from a branch, or do I need to make installers?

uli-2016, 2023-05-14:
Yes 600 ms sounds very long, it is the safety variants.
I have installed the version 5.3.6. I cannot compile, but I can replace files

bobjacobsen, 2023-05-14:
The updated code is in the `bobjacobsen/z21_POM_write_delay` branch. Installers are at:

https://builds.jmri.org/jenkins/job/testreleases/job/z21_pom_write_delay/55/

To change the delay value e.g.
```
Import jmri
jmri.jmrix.roco.z21.Z21XNetOpsModeProgrammer.operationDelay = 100
```

The default delay value is 250 milliseconds now.

I’m very interested in hearing if this change even works, as I have no way to test it.  So the first thing might be to set the value to e.g. 2000 and make sure that the delays are actually present and fix the problem.  Then it would be good to walk the time down to see what’s the minimum to fix the problem.

uli-2016, 2023-05-14:
additional: the index CV (eg. CV31 or CV1021) is set correctly when reading or writing an iCV. So the error occurs after writing the index CV and for the delay time.

bobjacobsen, 2023-05-14:
Thanks for the feedback.

Could you try the new installers at

https://builds.jmri.org/jenkins/job/testreleases/job/z21_pom_write_delay/56/

please? Thank you.

uli-2016, 2023-05-14:
Bob,
You have found the right button, 👍 with delay = 1000, read and write works.
in the next days I will make some tests with several decoders to find a shorter delay time

thanks

bobjacobsen, 2023-05-15:
Would it be useful to have separate timeout values for reads and writes?

uli-2016, 2023-05-15:
> Would it be useful to have separate timeout values for reads and writes?

No, after my tests, it is not neccessary. I suggest a "jmri.jmrix.roco.z21.Z21XNetOpsModeProgrammer.operationDelay = 50" for the z21 is good  default value. 
As long as only one locomotive is on the main track, everything works fine with POM. With several locomotives the RailCom protocol still has weaknesses. It comes thereby again and again to the error 301 (Programmer error: No locomotive detected). But this is a problem of the z21 and the decoders used. There are decoder types that interfere more than others.

With this modification for the delay time, my problem with reading iCVs with the z21 is now solved. The complete reading of decoders with index CVs is now very fast.

Thanks for the wounderful and fast support

bobjacobsen, 2023-05-15:
Fix is in PR #12025

closed 2023-05-17 by uli-2016