JMRI/JMRI #12012: Raildriver   .py file

Slank6000, 2023-05-10:
Hello. I have a Raildriver controller that I currently use with my model train (HO scale) via DCC-EX. I have a special raildriver.py file works fine on my layout system. I'm trying now to see if I can get the Raildriver to work on the new Windows OS 11, the latest realest of JMRI and DCC-EX. 
It Raildriver and Raildriver throttle works fine on my Windows 7 machine. I use it by simply running the raildriver.py file. This does not work with a newer 64 bit OS, etc. Any ideas? Thanks!

danielb987, 2023-05-10:
@Slank6000 
Welcome to JMRI!

I recommend that you ask on the JMRI users list. [Join](https://groups.io/g/jmriusers/join) and [Browse](https://groups.io/g/jmriusers/topics)

dsand47, 2023-05-10:
In what way does it "not work"?  

Start by posting the JMRI system console contents (Help => System Console, Copy to clipboard button).

Slank6000, 2023-05-10:
Hello and thanks!                              On my running layout. I launch Decoder Pro. Go to Run Script and load it. Then a throttle comes up and I canuse the JMRI throttle to select a loco. Loco address will then match the Raildriver's LED with the same loco address (unless long numbers).The I use the RailDriver console to control the loco or locos using the same functions (by number) as the JMRI throttle. If I have multiple locoslisted in the JMRI throttle, I can toggle through them on the RailDriver console to control each by loco address.
Right now when I run the raildriver.py ,,, it does nothing. I will pull Console Comments and paste them.
Thanks!!

Slank6000, 2023-05-10:
I thought I did. Okay. Thanks for the help!

Slank6000, 2023-05-10:
2023-05-10 17:34:53,955 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]2023-05-10 17:34:53,957 util.Log4JUtil                        INFO  - This log is appended to file: C:\Users\cgibs\JMRI\log\messages.log [main]2023-05-10 17:34:53,957 util.Log4JUtil                        INFO  - This log is stored in file: C:\Users\cgibs\JMRI\log\session.log [main]2023-05-10 17:34:53,985 apps.AppsBase                         INFO  - DecoderPro version 5.3.6+R7bfd8a5c79 starts under Java 11.0.18 on Windows 11 amd64 v10.0 at Wed May 10 17:34:53 EDT 2023 [main]2023-05-10 17:34:54,105 gui3.Apps3                            INFO  - Starting with profile My_JMRI_Railroad.3f4ca17b [main]2023-05-10 17:34:54,243 node.NodeIdentity                     INFO  - Using 3e81abfc-0acb-4a50-8c2e-da164f699bf9 as the JMRI storage identity for profile id 3f4ca17b [AWT-EventQueue-0]2023-05-10 17:34:54,300 xml.AbstractSerialConnectionConfigXml INFO  - Starting to connect for "DCC++" [main]2023-05-10 17:34:54,563 serial.DCCppAdapter                   INFO  - COM6 port opened at 115200 baud with DTR: true RTS: true DSR: false CTS: false  CD: false [main]2023-05-10 17:34:54,600 dccpp.DCCppInitializationManager      INFO  - DCC++ Initialization Complete with station type 'Unknown', version '0.0.0' and build 'Unknown' [main]2023-05-10 17:34:54,809 roster.Roster                         INFO  - Reading roster file with rootFromName(C:\Users\cgibs\JMRI\My_JMRI_Railroad.jmri\roster.xml) [main]2023-05-10 17:34:55,265 util.FileUtilSupport                  INFO  - File path program: is C:\Program Files (x86)\JMRI\ [main]2023-05-10 17:34:55,265 util.FileUtilSupport                  INFO  - File path preference: is C:\Users\cgibs\JMRI\My_JMRI_Railroad.jmri\ [main]2023-05-10 17:34:55,265 util.FileUtilSupport                  INFO  - File path profile: is C:\Users\cgibs\JMRI\My_JMRI_Railroad.jmri\ [main]2023-05-10 17:34:55,265 util.FileUtilSupport                  INFO  - File path settings: is C:\Users\cgibs\JMRI\ [main]2023-05-10 17:34:55,265 util.FileUtilSupport                  INFO  - File path home: is C:\Users\cgibs\ [main]2023-05-10 17:34:55,265 util.FileUtilSupport                  INFO  - File path scripts: is C:\Program Files (x86)\JMRI\jython\ [main]2023-05-10 17:35:02,722 jmrix.AbstractMRTrafficController     WARN  - Timeout on reply to message: Z consecutive timeouts = 0 in serial.SerialDCCppPacketizer [serial.SerialDCCppPacketizer Transmit thread]2023-05-10 17:35:03,812 dccpp.DCCppCommandStation             INFO  - Station Type set to 'DCC-EX' [AWT-EventQueue-0]2023-05-10 17:35:03,812 dccpp.DCCppCommandStation             INFO  - Build set to 'a26d988' [AWT-EventQueue-0]2023-05-10 17:35:03,812 dccpp.DCCppCommandStation             INFO  - Version set to '4.1.5' [AWT-EventQueue-0]2023-05-10 17:35:03,828 dccpp.DCCppCommandStation             INFO  - changing maxNumSlots from 12 to 50 [AWT-EventQueue-0]2023-05-10 17:35:33,001 input.DefaultControllerEnvironment    WARN  - Found unknown Windows version: Windows 11 [AWT-EventQueue-0]2023-05-10 17:35:33,001 input.DefaultControllerEnvironment    INFO  - Attempting to use default windows plug-in. [AWT-EventQueue-0]2023-05-10 17:35:33,001 input.DefaultControllerEnvironment    INFO  - Loading: net.bobis.jinput.hidraw.HidRawEnvironmentPlugin [AWT-EventQueue-0]java.lang.UnsatisfiedLinkError: no jhidrawplugin-win_64 in java.library.path: [., lib\x64, lib] at java.base/java.lang.ClassLoader.loadLibrary(Unknown Source) at java.base/java.lang.Runtime.loadLibrary0(Unknown Source) at java.base/java.lang.System.loadLibrary(Unknown Source) at net.bobis.jinput.hidraw.HidRawEnvironmentPlugin$2.run(HidRawEnvironmentPlugin.java:366) at java.base/java.security.AccessController.doPrivileged(Native Method) at net.bobis.jinput.hidraw.HidRawEnvironmentPlugin.loadLibrary(HidRawEnvironmentPlugin.java:358) at net.bobis.jinput.hidraw.HidRawEnvironmentPlugin.<clinit>(HidRawEnvironmentPlugin.java:89) at java.base/java.lang.Class.forName0(Native Method) at java.base/java.lang.Class.forName(Unknown Source) at net.java.games.input.DefaultControllerEnvironment.getControllers(DefaultControllerEnvironment.java:161) at jmri.jmrix.jinput.TreeModel.loadSystem(TreeModel.java:251) at jmri.jmrix.jinput.TreeModel.<init>(TreeModel.java:46) at jmri.jmrix.jinput.TreeModel.instance(TreeModel.java:98) at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method) at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(Unknown Source) at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source) at java.base/java.lang.reflect.Method.invoke(Unknown Source) at org.python.core.PyReflectedFunction.__call__(PyReflectedFunction.java:190) at org.python.core.PyReflectedFunction.__call__(PyReflectedFunction.java:208) at org.python.core.PyObject.__call__(PyObject.java:446) at org.python.core.PyObject.__call__(PyObject.java:450) at org.python.pycode._pyx0.f$0(<script>:317) at org.python.pycode._pyx0.call_function(<script>) at org.python.core.PyTableCode.call(PyTableCode.java:173) at org.python.core.PyCode.call(PyCode.java:18) at org.python.core.Py.runCode(Py.java:1687) at org.python.core.__builtin__.eval(__builtin__.java:497) at org.python.core.__builtin__.eval(__builtin__.java:501) at org.python.util.PythonInterpreter.eval(PythonInterpreter.java:255) at org.python.jsr223.PyScriptEngine.eval(PyScriptEngine.java:57) at org.python.jsr223.PyScriptEngine.eval(PyScriptEngin [... cut ...]        INFO  - Controller ITE Device(8910) has 3 components [AWT-EventQueue-0]2023-05-10 17:35:33,246 jinput.TreeModel                      INFO  - Controller RailDriver Modern Desktop has 32 components [AWT-EventQueue-0]2023-05-10 17:35:33,248 jinput.TreeModel                      INFO  - Controller ITE Device(8910) has 32 components [AWT-EventQueue-0]2023-05-10 17:35:33,248 jinput.TreeModel                      INFO  - Controller Controller (TGZ Controller) has 16 components [AWT-EventQueue-0]

bobjacobsen, 2023-05-10:
Thanks for the console.  It shows the problem:

2023-05-10 17:35:33,001 input.DefaultControllerEnvironment    WARN  - Found unknown Windows version: Windows 11 [AWT-EventQueue-0]
2023-05-10 17:35:33,001 input.DefaultControllerEnvironment    INFO  - Attempting to use default windows plug-in. [AWT-EventQueue-0]
2023-05-10 17:35:33,001 input.DefaultControllerEnvironment    INFO  - Loading: net.bobis.jinput.hidraw.HidRawEnvironmentPlugin [AWT-EventQueue-0]
java.lang.UnsatisfiedLinkError: no jhidrawplugin-win_64 in java.library.path: [., lib\x64, lib] 
at java.base/java.lang.ClassLoader.loadLibrary(Unknown Source) 
at java.base/java.lang.Runtime.loadLibrary0(Unknown Source) 
at java.base/java.lang.System.loadLibrary(Unknown Source) 
at net.bobis.jinput.hidraw.HidRawEnvironmentPlugin$2.run(HidRawEnvironmentPlugin.java:366)

The “net.bobis.jinput” library that JMRI uses to talk over the RailDriver USB link tried to access something in Windows called 'jhidrawplugin-win_64’ and it didn’t find it.

We distribute a number of Windows .dll files for jinput with JMRI:

lib/windows/x64/jinput-raw_64.dll
lib/windows/x86/jinput-dx8.dll
lib/windows/x86/jinput-raw.dll
lib/windows/x86/jinput-wintab.dll

I don’t know enough about Windows to suggest anything more. Perhaps somebody else does.

danielb987, 2023-05-10:
> It Raildriver and Raildriver throttle works fine on my Windows 7 machine. I use it by simply running the raildriver.py file. This does not work with a newer 64 bit OS, etc. Any ideas? Thanks!

Is your Windows 7 a 32 bit OS? The problem seems to be that you have a 32 bit `jhidrawplugin` and it won't work with 64 bit. Maybe you need to upgrade the Raildriver software?

Slank6000, 2023-05-12:
Thanks all for the help. The driver issue makes sense!!

closed 2023-05-12 by Slank6000