JMRI/JMRI #12018: Add Memories as a native MQTT object type

jerryg2003, 2023-05-12:
As per comments in various user groups and forums, there seems to be expanding interest in using MQTT as a layout control network protocol with JMRI.  We already have good MQTT support with "native" capability for sensors, turnouts, lights, reporters, signal head, signal mast.  I suggest that "Memories" be added.  [Note: MQTT messages can be directed to/from memories now via custom scripts..  This requires that the script be tailored to the specific memories designated to send/receive via MQTT.  This may be beyond the skill level of many users.]

1. Create a "Memory send topic" and "Memory receive topic" on the MQTT preferences page
2. Create an MQTT Memory Type for MQTT (prefixed as other objects are) . Right now all Memories are "internal" type only (prefix "I") so this would also entail changing the "Add New Memory" dialog to now have a System Connection pulldown that included any MQTT connections (perhaps more than one).
3. "Subscribe" to MQTT messages for any defined MQTT Memories whenever they are loaded from a panel or created.
4. Receive MQTT messages directed to the appropriate topic and put the free-form message content into the appropriate memory.
5. "Publish" MQTT messages whenever the contents of an MQTT Memory changes.

I realize this is likely a lot of work so may be worth polling the user community (jmriusers, MQTT4MRR, arduini, RFID4MRR, etc.) .  I  can, of course, test the code and update the documentation if/when this change is made.

bobjacobsen, 2023-05-12:
No other system has system-specific Memories.  

It’s usually the Reporter role to take in arbitrary input and make it available for various uses.

Is there some reason that a Reporter is insufficient, and a Memory is needed?

jerryg2003, 2023-05-12:
Using an MQTT Reporter results in the full message (including any spaces} being put in the Current Report field of an existing Reporter AND an IDTag being created (automatically if not already there) for the first part of the message before any spaces.  This is a specific function that works for RFID but is not particularly suited to the way I understood Memories are able to be used.  Further, other RFID protocols (MERG Concentrator and I believe LocoNet) will blank out the Current Report field either in 2 seconds or when the tag is no longer in view.  That’s OK for RFID tracking, but not for many other purposes.

For example, I use Memories to record a wide variety of information generated by my layout:  train speed, car count, IP address of the microcontroller (MC) publishing the message, direct messages that can be displayed on any type of JMRI panel, error messages from the MC, and who know what in the future.  I also “publish” (via a script) from Memories to send commands (such as “reboot” or “query” or “display on local display”).  I have these capabilities because I was able to write scripts to handle them.  I’m suggesting making these general purposes of Memories to anywho who can generate and/or receive an MQTT message.   It is a way of “future proofing” JMRI since we don’t have to make any decisions about what or how Memories are being used, as opposed to with Reporters.

danielb987, 2023-05-14:
I think a better solution is to add [AnalogIO](https://github.com/JMRI/JMRI/blob/master/java/src/jmri/AnalogIO.java) and [StringIO](https://github.com/JMRI/JMRI/blob/master/java/src/jmri/StringIO.java) support for MQTT to JMRI.

Example of AnalogIO support is meters, which are implemented in for example LocoNet. An example of StringIO is the [LnThrottleStringIO](https://github.com/JMRI/JMRI/blob/master/java/src/jmri/jmrix/loconet/LnThrottleStringIO.java) that sends a message to LocoNet throttles.

However, the implementation of AnalogIO and StringIO is currently limited so it would probably take a lot of work to get it work properly. For example, how do you create an AnalogIO for MQTT and configures it properly? When the AnalogIO value is published to MQTT, should it publish an integer value or a floating value, and if a floating value, how many decimals?

danielb987, 2023-05-14:
A related discussion on the MQTT4MR list. I add it here for reference.
https://groups.io/g/MQTT4MR/topic/98877595

closed 2023-07-15 by stale[bot]