JMRI/JMRI #12032: ESU function map: Description for Logical Functions broken (regression)

newHeiko, 2023-05-18:
As discussed on jmriusers:

Between 5.3.3 and 5.3.4, the ESU function map logical functions column lost their descriptions.

How to reproduce:
1. Create a roster entry for LokSound 5
2. Open (Program or Edit) the roster entry with Comprehensive Programmer
3. Switch to Function Map tab
4. On any of the lines, find the Logical Functions column and click on "Change" there

Working screenshot (from 5.2):
[image]

Broken screenshot (from master, i.e. 76de6b11a0):
[image]

Tracking this through git bisect returns (for me):

> There are only 'skip'ped commits left to test.
> The first bad commit could be any of:
> b35f463079ddadc50d04461a26fbf1e139440c72
> 8626a782c23a5552b626aa0c8ee382cd39fea408
> 9040ba149e0f5562baa647e8432b495097d72fad
> 05a07a0ac350fccddb2c8b5c0fe181b1c87da8c6
> We cannot bisect more! 

Heiko

newHeiko, 2023-05-18:
Addendum: Github tells me all those commits are from #11807 

Heiko

bobjacobsen, 2023-05-18:
See PR #12033

philguk, 2023-05-20:
WOW!

I was offering a sanity check to whoever picked this up to fix, but looks like it's fixed (and tested) already

Awesome job guys (especially @ bobjacobsen)

Probably time to refresh my JMRI donation :)

Oh, and I assume this will be in the next test release, or is there an interim install I can move to??

bobjacobsen, 2023-05-20:
It’ll be in the next test release in a couple weeks.

In the meantime, if you want to, you can pick up the most recent development build here:  https://builds.jmri.org/jenkins/job/development/job/packages/ 

This fix is already present there.

Bob

> On May 20, 2023, at 4:30 PM, philguk ***@***.***> wrote:
> Oh, and I assume this will be in the next test release, or is there an interim install I can move to??

—
Bob Jacobsen
***@***.***

philguk, 2023-05-21:
Downloaded and installed

Working as expected, thanks Bob

Hopefully you didn't have to reverse ALL the tidying up that had been done!

Phil G