JMRI/JMRI #12036: convert RosterExportToDCC-Ex.py script to JMRI Java

mstevetodd, 2023-05-19:
The RosterExportToDCC-Ex.py script populates a frame with the macro commands (text) needed to paste into DCC-EX's myAutomation.h so that the Roster Entries are available for throttle clients direct-connected to DCC-EX.
This feature would be much easier to access if it were part an option from the DCC++ menu, and as an action that could be added to a button from the Startup Preferences.
The logic should stay the same.
Remove the script when this feature is available, to avoid duplicate code and confusion.

danielb987, 2023-05-19:
A possibility is to add a menu item that runs the `RosterExportToDCC-Ex.py` script.

mstevetodd, 2023-05-22:
> 

@danielb987 , can you point to an example where where we have a script being called from a menu?

dsand47, 2023-05-22:
Steve,

Not a menu item, but you can a script button using a start up action. 

Dave Sand

danielb987, 2023-05-22:
> @danielb987 , can you point to an example where where we have a script being called from a menu?

No, but I can give you an example on how to call a script:

https://github.com/JMRI/JMRI/blob/master/java/src/jmri/jmrit/logixng/actions/ActionScript.java#L281

I haven't tested the code below, but I think this will be enough:
```
import java.io.*;
import java.nio.charset.StandardCharsets;
import javax.script.Bindings;
import javax.script.ScriptException;
import javax.script.SimpleBindings;
import jmri.script.ScriptEngineSelector;
```
```
private final ScriptEngineSelector _scriptEngineSelector = new ScriptEngineSelector();
```
```
Bindings bindings = new SimpleBindings();
ScriptEngineSelector.Engine engine =
            _scriptEngineSelector.getSelectedEngine();
if (engine == null) {
    throw new JmriException("Script engine is null");
}
try (InputStreamReader reader = new InputStreamReader(
            new FileInputStream(jmri.util.FileUtil.getExternalFilename(scriptFilename)),
            StandardCharsets.UTF_8)) {
    engine.getScriptEngine().eval(reader, bindings);
} catch (IOException | ScriptException e) {
    log.warn("cannot execute script \"{}\"", script, e);
}
```

I added some extra bindings in the variable `bindings` but I don't think you will need that so you could probably remove that variable.

commit 827c882f (2023-07-20): Merge pull request #12195 from mstevetodd/master

Call "Roster Export to DCC-EX" from DCC++ menu, closes #12036
commit f585fe3f (2023-07-20): Call "Roster Export to DCC-EX" from DCC++ menu, closes #12036
closed 2023-07-20 by danielb987