JMRI/JMRI #12037: ESU Decoder - unable to use programmer

Alan-Lomax, 2023-05-19:
Issue Type: __Bug__

I have an existing loco, that had a basic non sound decoder. Fully populated entry in the roster etc. I have just upgraded the decoder to an ESU V5. It runs around the track (on address 3) and responds to function keys. Ok so now I want to change the address and decoder in the roster. This does not seem possible via the UI.
As a workaround I created a new entry in the roster, selecting ESU V5 for decoder type. All attempts at reading/writing Cvs are resulting in error 303 command not implemented ... so i am stuck at the moment.

PanelPro 5.0+Rc441642522
OpenJDK 64-Bit Server VM Azul Systems, Inc. 11.0.16.1+1-LTS mixed mode
Windows 10 10.0 amd64

danielb987, 2023-05-20:
@Alan-Lomax 
I recommend that you ask on the JMRI users list. [Join](https://groups.io/g/jmriusers/join) and [Browse](https://groups.io/g/jmriusers/topics)

Alan-Lomax, 2023-05-23:
OK my issue is sorted now What I found was programming with an ops mode programmer simply did not work and resulted in the 303 error. (This was all on a yard of straight test track on the bench). When i moved the track feed over to the programming track outputs and used a programming track programmer then everything worked as intended and I was able to set address and adjust speed curves.

closed 2023-05-23 by Alan-Lomax