JMRI/JMRI #12062: LogixNG - Edit Contional - Add - Create button does not work

Aberdon, 2023-05-26:
Issue Type: __Bug__

Open 'LogixNG' table

Click 'Add'

Click 'Create'

On 'Edit LogixNG' window click 'New ConditionalNG'

On 'Add ConditionalNG' window click 'Create'

On 'Edir Conditional' window right click on female '!A' socket and click on 'Add - Item - Turnout'

Select an existing turnout ('Direct') and set it to be 'Closed' ('Direct')

Click 'Create' - nothing happens.

This is the same when trying to create any ConditionalNG, be it an if...then...else conditional, a sensor conditional, etc.  I have not been able to create any LogixNG ConditionalNG's!  Tried over several days, with fresh PC start-ups, only PanelPro running, etc - no luck.

Using PanelPro 5.2+R760b98537f with Java version 11.0.19(en) on a Windows 11 PC.

PanelPro 5.2+R760b98537f
OpenJDK 64-Bit Server VM OpenLogic 11.0.19+7-adhoc..jdk11u mixed mode
Windows 11 10.0 amd64

danielb987, 2023-05-26:
@Aberdon 
Welcome to the JMRI project!

When this error happens, go to Help => System Console to show the JMRI console.  Use the Copy to clipboard button and paste the clipboard in a reply.

Aberdon, 2023-05-26:
Here you go Dan, thanks for your help...

2023-05-26 19:04:32,686 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2023-05-26 19:04:32,691 util.Log4JUtil                        INFO  - This log is appended to file: C:\Users\ianmu\JMRI\log\messages.log [main]
2023-05-26 19:04:32,692 util.Log4JUtil                        INFO  - This log is stored in file: C:\Users\ianmu\JMRI\log\session.log [main]
2023-05-26 19:04:32,752 apps.Apps                             INFO  - PanelPro version 5.2+R760b98537f starts under Java 11.0.19 on Windows 11 amd64 v10.0 at Fri May 26 19:04:32 BST 2023 [main]
2023-05-26 19:04:32,881 apps.Apps                             INFO  - Starting with profile My_JMRI_Railroad.3e52f892 [main]
2023-05-26 19:04:33,139 node.NodeIdentity                     INFO  - Using ca659cb5-4ea4-4dab-b799-91cfb50d0195 as the JMRI storage identity for profile id 3e52f892 [AWT-EventQueue-0]
2023-05-26 19:04:33,478 mqtt.MqttAdapter                      INFO  - MQTT starting connect with MQTTchannel = "" [main]
2023-05-26 19:04:35,262 util.FileUtilSupport                  INFO  - File path program: is C:\Program Files (x86)\JMRI\ [main]
2023-05-26 19:04:35,262 util.FileUtilSupport                  INFO  - File path preference: is C:\Users\ianmu\JMRI\My_JMRI_Railroad.jmri\ [main]
2023-05-26 19:04:35,262 util.FileUtilSupport                  INFO  - File path profile: is C:\Users\ianmu\JMRI\My_JMRI_Railroad.jmri\ [main]
2023-05-26 19:04:35,262 util.FileUtilSupport                  INFO  - File path settings: is C:\Users\ianmu\JMRI\ [main]
2023-05-26 19:04:35,262 util.FileUtilSupport                  INFO  - File path home: is C:\Users\ianmu\ [main]
2023-05-26 19:04:35,263 util.FileUtilSupport                  INFO  - File path scripts: is C:\Program Files (x86)\JMRI\jython\ [main]
2023-05-26 19:04:36,623 PanelPro.PanelPro                     INFO  - Main initialization done [main]
2023-05-26 19:04:43,821 configurexml.LoadXmlConfigAction      INFO  - Loading selected file: C:\Users\ianmu\JMRI\My_JMRI_Railroad.jmri\Test.xml [AWT-EventQueue-0]
2023-05-26 19:04:53,240 audio.JoalAudioFactory                INFO  - Initialised JOAL using OpenAL: vendor - OpenAL Community version - 1.1 ALSOFT 1.15.1 [Listed Table Generation]
2023-05-26 19:05:12,468 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [JMRI LogixNGThread]
java.lang.IllegalArgumentException: system name is not valid: IQDA1
	at jmri.jmrit.logixng.actions.AbstractDigitalAction.<init>(AbstractDigitalAction.java:36)
	at jmri.jmrit.logixng.actions.ActionTurnout.<init>(ActionTurnout.java:30)
	at jmri.jmrit.logixng.actions.swing.ActionTurnoutSwing.validate(ActionTurnoutSwing.java:73)
	at jmri.jmrit.logixng.tools.swing.TreeEditor.lambda$createAddFrame$9(TreeEditor.java:495)
	at jmri.jmrit.logixng.util.LogixNG_Thread.lambda$new$0(LogixNG_Thread.java:143)
	at java.base/java.lang.Thread.run(Unknown Source)

I have also noticed that if I try to create a 'MQTT' - 'Publish' ConditionalNG then nothing happens and the following is shown on the System Console...

2023-05-26 19:06:51,300 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
java.lang.IllegalArgumentException: system name is not valid: IQDA1
	at jmri.jmrit.logixng.actions.AbstractDigitalAction.<init>(AbstractDigitalAction.java:36)
	at jmri.jmrit.logixng.actions.AbstractDigitalAction.<init>(AbstractDigitalAction.java:26)
	at jmri.jmrix.mqtt.logixng.Publish.<init>(Publish.java:34)
	at jmri.jmrix.mqtt.logixng.swing.PublishSwing.createPanel(PublishSwing.java:43)
	at jmri.jmrit.logixng.actions.swing.AbstractDigitalActionSwing.getConfigPanel(AbstractDigitalActionSwing.java:69)
	at jmri.jmrit.logixng.tools.swing.TreeEditor.makeAddEditFrame(TreeEditor.java:796)
	at jmri.jmrit.logixng.tools.swing.TreeEditor.createAddFrame(TreeEditor.java:560)
	at jmri.jmrit.logixng.tools.swing.TreeEditor$PopupMenu.lambda$addNewItemTypes$1(TreeEditor.java:1941)
	at java.desktop/javax.swing.AbstractButton.fireActionPerformed(Unknown Source)
	at java.desktop/javax.swing.AbstractButton$Handler.actionPerformed(Unknown Source)
	at java.desktop/javax.swing.DefaultButtonModel.fireActionPerformed(Unknown Source)
	at java.desktop/javax.swing.DefaultButtonModel.setPressed(Unknown Source)
	at java.desktop/javax.swing.AbstractButton.doClick(Unknown Source)
	at java.desktop/javax.swing.plaf.basic.BasicMenuItemUI.doClick(Unknown Source)
	at java.desktop/javax.swing.plaf.basic.BasicMenuItemUI$Handler.mouseReleased(Unknown Source)
	at java.desktop/java.awt.Component.processMouseEvent(Unknown Source)
	at java.desktop/javax.swing.JComponent.processMouseEvent(Unknown Source)
	at java.desktop/java.awt.Component.processEvent(Unknown Source)
	at java.desktop/java.awt.Container.processEvent(Unknown Source)
	at java.desktop/java.awt.Component.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Component.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.LightweightDispatcher.retargetMouseEvent(Unknown Source)
	at java.desktop/java.awt.LightweightDispatcher.processMouseEvent(Unknown Source)
	at java.desktop/java.awt.LightweightDispatcher.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.Container.dispatchEventImpl(Unknown Source)
[... cut ...]
	at java.desktop/java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.run(Unknown Source)

dsand47, 2023-05-26:
It looks like you are trying to create you own LogixNG system names.  

I recommend use only auto generated system names.

Dave Sand

at jmri.jmrit.logixng.actions.AbstractDigitalAction.(AbstractDigitalAction.java:36)
at jmri.jmrit.logixng.actions.ActionTurnout.(ActionTurnout.java:30)

at jmri.jmrit.logixng.actions.AbstractDigitalAction.(AbstractDigitalAction.java:36)
at jmri.jmrit.logixng.actions.AbstractDigitalAction.(AbstractDigitalAction.java:26)
at jmri.jmrix.mqtt.logixng.Publish.(Publish.java:34)

danielb987, 2023-05-26:
@Aberdon @dsand47 
There is a bug in LogixNG. If the internal connection doesn't have the name `I` (uppercase `i`), the creation of LogixNGs fail.

I wrongly assumed that there always exists an internal connection with the name `I`.

bobjacobsen, 2023-05-26:
There really should be an "I" system, as it's automatically created and can't be changed by the user.  

Do we have any idea why it's missing in this case?

dsand47, 2023-05-26:
Bob,

If a None connection is defined, it uses I2 and I disappears.  After restarting it is possible to change the connection prefix to just I.

Dave Sand

bobjacobsen, 2023-05-26:
Thanks!

Sounds like the original “I” connection disappearing is a bug.  I’ll try to track it down.

dsand47, 2023-05-26:
Bob,

If "None" is not first connection, the regular internal works and Internal2 is not available..  This implies that the issue might be related the "default connection" logic.

Unfortunately "None" is used when no layout connection is planned or is undecided.

Dave Sand

danielb987, 2023-05-26:
@bobjacobsen 
> If "None" is not first connection, the regular internal works and Internal2 is not available.. This implies that the issue might be related the "default connection" logic.
> 
> Unfortunately "None" is used when no layout connection is planned or is undecided.

There is a potential problem if this bug is fixed. If someone created a "None" connection and got the name "I2" and then renamed it to "I", what happens when this bug is fixed? Will that "None" connection with the name "I" fail to load?

dsand47, 2023-05-26:
Daniel,

Maybe.

The other potential issue is the people who have been using "None" and their internal objects are currently using I2 as the system name prefix.

Dave Sand

Aberdon, 2023-05-26:
> It looks like you are trying to create you own LogixNG system names. I recommend use only auto generated system names. Dave Sand
> 
No, I am leaving all settings as they default - the "LogixNg System Name" is set to "Automatically generate System Name".

dsand47, 2023-05-26:
Is your layout connection set to "None"?  If so, change to something like Digitrax simulator.

Dave Sand

Aberdon, 2023-05-26:
> Is your layout connection set to "None"? If so, change to something like Digitrax simulator. Dave Sand

That is it!  Thank you.

I have been using a test panel with just a MQTT connection.  I have now set up a DCC++ Simulator connection too and all is now working, I can create ConditionalNG's for everything I have tried so far.

Thank you once again 👍

closed 2023-07-27 by stale[bot]