JMRI/JMRI #12091: Script to Initialize all physical Turnouts and Sensors

Cliff-Anderson, 2023-06-02:
At the  [suggestion](https://groups.io/g/jmriusers/message/218105) of @PUlvestad, this "InitAllTurnoutsSensors" script to initialize all known physical Turnouts to CLOSED and all known physical Sensors to INACTIVE is offered for your consideration.  

The earliest version was developed for the startup at our club layout and some variation is still in use there.  

Also following Peter's suggestion, some time has be taken to edit and revise both the code and the comments text to conform to the JMRI version 5.3.6 context.  

Previous versions of the script have been in the [files/Script Examples](https://groups.io/g/jmriusers/files/Script%20examples) folder since about 2014, but I am unaware of how many times it might have been downloaded, or if even once.  

[InitAllTurnoutsSensors.zip](https://github.com/JMRI/JMRI/files/11638177/InitAllTurnoutsSensors.zip)

dsand47, 2023-06-02:
You should use getSystemName() and not rely on the implied toString() to return the only the system name.

Cliff-Anderson, 2023-06-03:
There are many ways to do the same thing in software.

What is the most convenient way to do the replacement to use getSystemName() instead of toString()?  

Should a new copy with a different file name be uploaded, or will the developers' normal review process take the note from @dsand47 into their editing?

danielb987, 2023-06-03:
@Cliff-Anderson 
It's good if you could do the change and then upload the new script. That way, you can verify that the change works as expected.

Cliff-Anderson, 2023-06-03:
Modified and verified script has same name but different zip filename.  Now uses getSystemName() in both occurrences.  No other changes.

[2nd InitAllTurnoutsSensors.zip](https://github.com/JMRI/JMRI/files/11641716/2nd.InitAllTurnoutsSensors.zip)

danielb987, 2023-06-04:
@dsand47 
Is this script ready for a PR?

dsand47, 2023-06-04:
> @dsand47
> Is this script ready for a PR?

I have not tested it but it looks OK.

danielb987, 2023-06-04:
See PR #12094.

closed 2023-06-05 by danielb987