JMRI/JMRI #12093: Dispatch with JMRI Throttle to FRED(I) doesn't work (ex-#9607 - closed)

PinoDiB, 2023-06-04:
Hello! I don't know if this issue - related to JMRI-DR5000 - has already been fixed, however I cannot dispatch a loco from JMRI throttle to FREDI while it is possible when the dispatch action starts from Digitrax UT4 instead of JMRI throttle. No issues when I try to dispatch directly from DR5000 throttle using Digikeijs software instead of JMRI. I can provide log file.

bobjacobsen, 2023-06-04:
it would be helpful if you could provide a LocoNet Monitor log file of the successful dispatch using the DR5000 throttle, and a second LocoNet Monitor log of the unsuccessful attempt with JMRI.  With raw data and timestamps please. Thanks in advance.

Pugwash1, 2023-06-04:
I suspect that this is related to reading the slot after it has been sent to another status, digitrax command station do not report that a slot sent to common infact has been sent idle or even ignored. This maybe confusing the non digitrax CS which though it uses  loconet is not actually  fully compatible to a Digitrax command station.

PinoDiB, 2023-06-05:
> it would be helpful if you could provide a LocoNet Monitor log file of the successful dispatch using the DR5000 throttle, and a second LocoNet Monitor log of the unsuccessful attempt with JMRI. With raw data and timestamps please. Thanks in advance.

Thank you, Bob.
This file is related to the dispatch made through Digikeijs software and recorded on JMRI Loconet monitor:
[DR5000SW.txt](https://github.com/JMRI/JMRI/files/11651231/DR5000SW.txt) .

In this one the dispatch is sent from JMRI throttle (unsuccessful):
[FREDIUnsuccessful.txt](https://github.com/JMRI/JMRI/files/11651305/FREDIUnsuccessful.txt) .

JMRI - Kindly note that when the dispatch starts from Digitrax UT4 instead of JMRI throttle, FREDI get the dispatch correctly.

PinoDiB, 2023-06-05:
> I suspect that this is related to reading the slot after it has been sent to another status, digitrax command station do not report that a slot sent to common infact has been sent idle or even [...]

Perhaps... I saw that, when the dispatch starts from JMRI throttle, the slot state changes in "common" and Fredi is unable to get it.

PinoDiB, 2023-06-05:
> I suspect that this is related to reading the slot after it has been sent to another status, digitrax command station do not report that a slot sent to common infact has been sent idle or even [...]

Additionally when the slot is in "common" state after the dispatch put and  I use another throttle, the Uhlenbrock Fred (a device similar to FREDI), it can get dispatch successfully.

PinoDiB, 2023-06-07:
Thank you, Bob! 
At the moment, I have no other Loconet Command Stations. In the next weeks I hope I will get a Digitrax DCS52 (Digitrax items are not easily available here), since Fremo guys don't know any issues relative to FREDI last generation (like mine) connected to Command Stations different from DR5000. 

I synthesize what happens in the attached file
[FrediTab.pdf](https://github.com/JMRI/JMRI/files/11678625/FrediTab.pdf)

PinoDiB, 2023-07-22:
Hello! Bad news. I tried to dispatch a loco through my Digitrax DCS52 using Fremo FREDI or Uhlenbrock Fred without results. Nothing changed using JMRI. Additionally, it was not possible to dispatch a loco from Digitrax UT4 too (it was possible when I used Digikeijs/YaMoRC command station). Digitrax Help Desk provided no help since the throttles are manufactured by other companies. 
[DCS52test.txt](https://github.com/JMRI/JMRI/files/12136136/DCS52test.txt)

danielb987, 2023-07-22:
Try to dispatch to the Fremo FREDI using only the Digitrax DCS52. Don't have anything else connected to the DCS52.

I'm not sure, but I think you dispatch from the DCS52 by:
Press Loco button
Enter loco number
Press Exit button
Press Loco button
Press Exit button

PinoDiB, 2023-07-22:
Hi Daniel! I followed the path described in Digitrax DCS52 Manual: 1. press Loco key; 2. press C key "MORE"; press A key "DISPATCH" - without other connections or JMRI - getting no result. Then, in order to watch better the changes in JMRI Slot Monitor I selected an address first (please see the attached file).
[DCS52FREDI.txt](https://github.com/JMRI/JMRI/files/12136749/DCS52FREDI.txt)

Pugwash1, 2023-07-22:
Loco
Enter number eg 1234
Loco To Aquire it.
Loco To release it.
More -> dispatch it.

16:53:33.911: [BF 09 52 1B]  Request slot for loco address 1234.
16:53:33.922: [E7 0E 09 03 52 00 00 47 00 09 00 00 00 00]  Report of slot 9 information:
	Loco 1234 is Not Consisted, Free, operating in 128 SS mode, and is moving Forward at speed 0,
	F0=Off, F1=Off, F2=Off, F3=Off, F4=Off, F5=Off, F6=Off, F7=Off, F8=Off
	Master supports LocoNet 1.1; Track Status: On/Running; Programming Track Status: Available; STAT2=0x00, ThrottleID=0x00 0x00 (0).
16:53:36.936: [BA 09 09 45]  Set status of slot 9 to IN_USE.
16:53:36.945: [E7 0E 09 33 52 00 00 47 00 09 00 00 00 30]  Report of slot 9 information:
	Loco 1234 is Not Consisted, In-Use, operating in 128 SS mode, and is moving Forward at speed 0,
	F0=Off, F1=Off, F2=Off, F3=Off, F4=Off, F5=Off, F6=Off, F7=Off, F8=Off
	Master supports LocoNet 1.1; Track Status: On/Running; Programming Track Status: Available; STAT2=0x00, ThrottleID=0x00 0x00 (0).
16:53:36.965: [EF 0E 09 33 52 00 00 47 00 09 00 6C 29 7D]  Write slot 9 information:
	Loco 1234 is Not Consisted, In-Use, operating in 128 SS mode, and is moving Forward at speed 0,
	F0=Off, F1=Off, F2=Off, F3=Off, F4=Off, F5=Off, F6=Off, F7=Off, F8=Off
	Master supports LocoNet 1.1; Track Status: On/Running; Programming Track Status: Available; STAT2=0x00, ThrottleID=0x29 0x6C (5356).
16:53:36.965: [B4 6F 7F 5B]  LONG_ACK: Function not implemented, no reply will follow.
16:53:36.965: [A0 09 00 56]  Set speed of loco in slot 9 to 0.
16:53:42.068: [B5 09 13 50]  Write slot 9 with status value 19 (0x13) - Loco is Not Consisted, Common and operating in 128 speed step mode.
16:53:45.720: [BA 09 00 4C]  Mark slot 9 as DISPATCHED.
16:54:06.073: [BA 00 00 45]  Get most recently dispatched slot.
16:54:06.083: [E7 0E 09 33 52 00 00 47 00 09 00 6C 29 75]  Report of slot 9 information:
	Loco 1234 is Not Consisted, In-Use, operating in 128 SS mode, and is moving Forward at speed 0,
	F0=Off, F1=Off, F2=Off, F3=Off, F4=Off, F5=Off, F6=Off, F7=Off, F8=Off
	Master supports LocoNet 1.1; Track Status: On/Running; Programming Track Status: Available; STAT2=0x00, ThrottleID=0x29 0x6C (5356).

PinoDiB, 2023-07-31:
Thank you!
Perhaps I am missing something:

12:54:26.601: [BF 09 52 1B]  Request slot for loco address 1234.
12:54:26.612: [E7 0E 03 13 52 00 00 46 00 09 00 3C 54 73]  Report of slot 3 information:
	Loco 1234 is Not Consisted, Common, operating in 128 SS mode, and is moving Forward at speed 0,
	F0=Off, F1=Off, F2=Off, F3=Off, F4=Off, F5=Off, F6=Off, F7=Off, F8=Off
	Master supports LocoNet 1.1; Track Status: Off/Running; Programming Track Status: Available; STAT2=0x00, ThrottleID=0x54 0x3C (10812).
12:54:27.658: [BB 02 00 46]  Request data/status for slot 2.
12:54:27.668: [E7 0E 02 33 77 00 10 46 00 01 00 71 02 74]  Report of slot 2 information:
	Loco 247 is Not Consisted, In-Use, operating in 128 SS mode, and is moving Forward at speed 0,
	F0=On, F1=Off, F2=Off, F3=Off, F4=Off, F5=Off, F6=Off, F7=Off, F8=Off
	Master supports LocoNet 1.1; Track Status: Off/Running; Programming Track Status: Available; STAT2=0x00, ThrottleID=0x02 0x71 (369).
12:54:28.182: [BA 03 03 45]  Set status of slot 3 to IN_USE.
12:54:28.192: [E7 0E 03 33 52 00 00 46 00 09 00 3C 54 53]  Report of slot 3 information:
	Loco 1234 is Not Consisted, In-Use, operating in 128 SS mode, and is moving Forward at speed 0,
	F0=Off, F1=Off, F2=Off, F3=Off, F4=Off, F5=Off, F6=Off, F7=Off, F8=Off
	Master supports LocoNet 1.1; Track Status: Off/Running; Programming Track Status: Available; STAT2=0x00, ThrottleID=0x54 0x3C (10812).
12:54:28.201: [EF 0E 03 33 52 00 00 46 00 09 00 3C 54 5B]  Write slot 3 information:
	Loco 1234 is Not Consisted, In-Use, operating in 128 SS mode, and is moving Forward at speed 0,
	F0=Off, F1=Off, F2=Off, F3=Off, F4=Off, F5=Off, F6=Off, F7=Off, F8=Off
	Master supports LocoNet 1.1; Track Status: Off/Running; Programming Track Status: Available; STAT2=0x00, ThrottleID=0x54 0x3C (10812).
12:54:28.205: [B4 6F 7F 5B]  LONG_ACK: Function not implemented, no reply will follow.
12:54:28.209: [A0 03 00 5C]  Set speed of loco in slot 3 to 0.
12:54:43.573: [A1 03 00 5D]  Set loco in slot 3 direction Forward F0=Off F1=Off F2=Off F3=Off F4=Off.
12:54:43.786: [B5 03 13 5A]  Write slot 3 with status value 19 (0x13) - Loco is Not Consisted, Common and operating in 128 speed step mode.
12:54:43.895: [BB 03 00 47]  Request data/status for slot 3.
12:54:43.905: [E7 0E 03 13 52 00 00 46 00 09 00 3C 54 73]  Report of slot 3 information:
	Loco 1234 is Not Consisted, Common, operating in 128 SS mode, and is moving Forward at speed 0,
	F0=Off, F1=Off, F2=Off, F3=Off, F4=Off, F5=Off, F6=Off, F7=Off, F8=Off
	Master supports LocoNet 1.1; Track Status: Off/Running; Programming Track Status: Available; STAT2=0x00, ThrottleID=0x54 0x3C (10812).
12:54:45.772: [BA 03 00 46]  Mark slot 3 as DISPATCHED.
12:54:51.995: [BA 00 00 45]  Get most recently dispatched slot.
12:54:52.000: [B4 3A 00 71]  LONG_ACK: The Move Slots command was rejected.

Pugwash1, 2023-07-31:
Ah, La Differnce.
Dr cs has ba 03 as dispatch digitrax has ba 09.
Get as far as sending it to common and rhen use the send loconet message to see if the ba 09 works better.

Pugwash1, 2023-08-01:
Ignore that. You were dispatching slot 3 I was dispatching slot 9..
Is that last trace from the dr5000 or dcs52?

PinoDiB, 2023-08-01:
Thank you! It is a Digitrax DCS52 trace.

Pugwash1, 2023-08-01:
@PinoDiB 
I cannot duplicate the problem using my dcs52. Do you know the firmware version?

PinoDiB, 2023-08-01:
@Pugwash1  
SW ver: 0000.1
(SubV 0.7 Jan23, 2023)

Pugwash1, 2023-08-02:
@PinoDiB 
You need the previous version  March21? The latest and greatest Jan23 doesnt appear to support dispatching.
Someone needs to tell digitrax, Im not too sure were my cred stands with them at the moment but will give it a try.

PinoDiB, 2023-08-04:
@Pugwash1 
If I understand well your words: my Digitrax DCS52 cannot dispatch a loco and this issue is related to the firmware, because your DCS52 - equipped with an older firmware - is able to dispatch successfully a loco. Looking at my Loconet Monitor,  the slot is marked as dispatched but it is impossible to complete the action (GET) because it is rejected.

I am blighted. My intention was to check if Fredi could get dispatch through a Command Station different from my Digikeijs/YaMoRC CS, as per Bob advice. I am back again to step number 1!

closed 2023-12-05 by github-actions[bot]