JMRI/JMRI #12119: Default Preferences are Invalid

rob-apps, 2023-06-19:
Hi - I updated to the latest production release, and I keep getting an error message that says "The Defaults preferences are invalid." "Do you want to correct them?"  If I select yes, it takes me to the Defaults settings page where I can select different connection default options.  No matter how many times I go through this process and no matter what I select, every single time I re-open JMRI, I get a new error message for the same thing.

I also noticed that when I try to read a decoder (Broadway-Limited), I get a "no acknowledgement from decoder error."  I'm not sure if that is related or unrelated to the Defaults error described above.  

I'm using a Digitrax PR4 as my interface on a Mac.  I didn't have this issue on my prior JMRI version.  Thanks in advance for any help.

bobjacobsen, 2023-06-19:
Have you saved the new preferences?

rob-apps, 2023-06-19:
Yes, and after saving it asks to restart, which I do.

bobjacobsen, 2023-06-19:
Here's what your defaults should probably look like:

[image]

LocoNet all across the line.  Is that what you have?

If so, something has gone wrong in the storage of the preferences.  Not sure what.  

Do you use profiles?  If so, a workaround might be to create a new one from scratch.

rob-apps, 2023-06-19:
I'm not sure my last response went through.  I attached some pictures, but maybe the file size was too big.  I can select all of the bubbles except the Command station one, which is greyed out.  Then I click the save icon in the lower left corner, at which point it asks me to restart, which I do.  Then, when it starts back up, all of the bubbles are empty again, so my choices never get saved.  I then get an error message that my defaults are not selected.  Should I just completely delete the software from my computer and reinstall?  If I go that route, does exporting my roster allow me to not have to manually input my locos again?  Thanks.

bobjacobsen, 2023-06-19:
(Attachments to emails don't come through; you have to use the web interface to attach images)

Being unable to select the LocoNet command station is probably what's causing the problem.  I don't know why that would happen.

Deleting JMRI won't help.  When you reinstall, you'll get the same program back.  The problem is in your preferences.

What we need to do is to clear our your preferences without losing your roster.  But first it would be good to understand what's going wrong. Let's try this:
1) First, back up your roster with "Export Complete Roster" from the file menu.  Put the resulting file somewhere safe.
2) From Help -> File Locations, click on "Open Profile Location".  There will be a "profile" directory there.  Right click to make a compressed version, and attach it to this PR (or mail it to directly me, rgj1927@gmail.com)

rob-apps, 2023-06-19:
Okay, I e-mailed the compressed profile to you directly.  Appreciate your help.

bobjacobsen, 2023-06-19:
Thanks.  Got it. It shows that you have two LocoNet connections:  One to your standalone PR4 and another to a DCS240 command station.  Is that right?

rob-apps, 2023-06-19:
Because I was having problems with the BLI Paragon 3 decoder not responding to read/write commands from the PR4, I switched to the USB input on my Digitrax 210+ to see if that would fix the problem. So I set up a second connection for that. But I only have one USB cable, so I can either connect through the PR4 or I can connect to the 210+, but not both at the same time.

bobjacobsen, 2023-06-19:
Did the defaults problem exist before you created the 2nd connection?  Or was that the problem you were trying to solve?

Try deleting one of those connections. To do that, click the little x box on the tab for the connection you don't want. Save and restart.

If you want have two different configurations, that's what profiles are meant for.  [https://www.jmri.org/help/en/html/setup/profiles.shtml](https://www.jmri.org/help/en/html/setup/profiles.shtml)

rob-apps, 2023-06-19:
No, it existed before I added the second one. I added the 210+ one because I was getting “no acknowledgment from decoder” errors when I was using the PR4 to try and program a BLI decoder. So I decided to try programming through the 210+. The decoder issue may be that I need a booster, like a PTB100 or the one made by DCC specialties. I’ve been reading that BLI decoders can be finicky. But the answer is that it was not saving the defaults even when my only connection was the PR4.

bobjacobsen, 2023-06-19:
OK, let's go back to trying to clear out your preferences.
1) Make that backup of your roster and make sure it's stored away safely.
2) open the location and drag the "8d07ae52-4eb0-43b2-b332-37d59967bc58" directory there to the trash.
3) Stop JMRI
4) Take the attached profile.xml.txt file (you might have to get it from the GItHub web site if it's not attached to the email version), rename it as profile.xml, and put it in place of the profile.xml file in the profile directory.
5) Restart JMRI and see whether it still prompts for defaults.

[profile.xml.txt](https://github.com/JMRI/JMRI/files/11793899/profile.xml.txt)

rob-apps, 2023-06-20:
Unfortunately that didn't work, and the weird thing is that the 8d07ae52... folder reappears in the profile folder after I open JMRI back up, even though I definitely sent it to the trash bin. Also, the profile folder already contains a file named "profile.xml," so when I put the new profile.xml in the profile directory, it asks me if I want to replace the existing one or keep both.

bobjacobsen, 2023-06-21:
Resolved offline.

closed 2023-06-21 by bobjacobsen