JMRI/JMRI #12129: Connect JMRI on MAC with Apple processor to DCS 240 using USB

David-Toll, 2023-06-27:
I have an iMac with the Apple M1 processor, running the latest version of MACOS Ventura

I have Installed Java 11 on this machine

I have downloaded JMRI 5.2 for the Apple processor MACs.

When I start up Decoder Pro it asks some questions:  I selected:
    Digitrax
   DCS240 USB Interfac3e

The next question is "Serial port".  However, as is stated in at least one of your web pages, no modern MACs have a serial port.

The Web page "**Connecting JMRI to a Digitrax DCS240**" says in its first sentence:

"JMRI may be connected to the Digitrax DCS240 in any of a wide variety of ways. Since the DCS240 has a built-in USB interface, it is possible to simply plug a USB cable into the computer and the DCS240, and configure JMRI to suit this connection."

When I connect the iMac and DCS 24 using a USB-A -> USB-B cable, and on the iMac look in "Preferences -> General -> About -> System Report -> USB, a device "VCOM DCS240" is listed.  This device disappears when the USB cable is unplugged from the iMac

I conclude from this that the iMac recognizes the presence of and removal of the DCS240.

However, as far as I can see,  Decoder Pro provides no means of connecting to this USB device.

How do I make it work?

Thank You for your help

David Toll

toll.david.c@gmail.com

dsand47, 2023-06-27:
David,

Look at https://www.jmri.org/install/MacOSX.shtml#debug

Pay attention to the section on identifying the device name.  You might need to install a driver.

Dave Sand

bobjacobsen, 2023-06-27:
On my PowerBook Pro, a DCS210 (not exactly the same, but close) native adapter shows up in the JMRI configuration selector as "cu.usbmodem2100412E1".  Do you see a choice like that?  The number will differ, as that's the serial number of the specific device.

David-Toll, 2023-06-27:
I do have a device "cu.usbmodemDtrx30DA1".  I tried this.

The "Connection Prefix" is set to "L"
The "Connection Name" is set to :LocoNet"

It then fails with: "Unable to create connection "LocoNet" (L).

If I unplug the USB cable, the device device "cu.usbmodemDtrx30DA1" is then shown in red

David-Toll, 2023-06-27:
I have tried the following command, as specified on the web page "JMRI Install Guide: macOS"

while : ;do clear;ls -lt /dev|head;i=$((i+1));echo $i;sleep 1;done

When the USB cable is connected, the above command shows two devices:
    cu.usbmodemDtrx30DA1
    tty.usbmodemDtrx30DA1

These devices vanish if I unplug the USB cable.

It was suggested above I may need a drive.  Does anyone know what driver, and where to I get it from?

thanks

bobjacobsen, 2023-06-27:
You'e got the right device now.  Something else is going wrong. 

Did you download the Apple Silicon or Intel version of Java?  If it was Apple Silicon, try the Intel one.  Rosseta2 will let you use that, and it seems to be more stable.

dheap, 2023-06-28:
David,

> On 28 Jun 2023, at 6:16 am, David-Toll ***@***.***> wrote:
> 
> When the USB cable is connected, the above command shows two devices:
> cu.usbmodemDtrx30DA1
> tty.usbmodemDtrx30DA1

Always pick the "cu.…" option on a Mac. The "tty.…" option won't work on Mac.

Dave in Australia

David-Toll, 2023-06-28:
I am using the CU form.I obtained my Java 11 from Oracle.  However I have been told that this version does not work well, and I should get it from Azul.  I cannot find any JRE or JDK downloads there.We’re did you get your Java from?ThanksDavid tollSent from my iPad

David,

Always pick the "cu.…" option on a Mac. The "tty.…" option won't work on Mac.

Dave in Australia

bobjacobsen, 2023-06-28:
Try this link:

https://www.azul.com/downloads/?version=java-11-lts&amp;package=jre

dsand47, 2023-06-28:
David,

Go to https://www.azul.com/downloads/?package=jdk#zulu

Scroll down a bit.

Select each category.
Use the JDK DMG download.  That is the easiest.  For Mx Macs, you can use either x86/64 or ARM64.  For some JMRI configurations, the Intel version works better.

Since you already have another install, you will need to go to /Library/Java/JavaVirtualMachines and remove the previous install.  This requires admin authority.

Dave Sand

bobjacobsen, 2023-06-28:
>  For Mx Macs, you can use either x86/64 or ARM64.  For some JMRI configurations, the Intel version works better.

With the error you're seeing, I recommend the x86/64 version.

David-Toll, 2023-06-28:
Thanks for the info From where do you suggest I get the JDK download?The current one I o rained from Oracle, but it has been suggested that the Oracle version does not work too wellThanksDavid

David, 

Always pick the "cu.…" option on a Mac. The "tty.…" option won't work on Mac. 

Dave in Australia

dsand47, 2023-06-28:
David,

The first line in my previous reply has the URL.  Copy the URL to your browser address field.  After making the selections, click on the ".dmg" link under the Download button.

Dave Sand

David-Toll, 2023-06-29:
Thanks to al who have responded and made suggestions.

(1) I power cycled my iMAC (i do not know if that made any difference, but after decades of using computers, a power cycle is one of the things to try!)

(2) I went to https://www.azul.com/downloads/?package=jdk#zulu

(3) as recommend above, I went to to https://www.azul.com/downloads/?package=jdk#zulu and downloaded 
    zulu11.64.16-ca-jdk11.0.19-macosx_x64.dmg

(4) I deleted the old Java Library, then double clicked the dmg file to install it.

Then, when I connected the iMac to the DCS 240, with a loco on the programming track, and started DecoderPro, it came straight up (without going through the initial programmer selection), and it seems to work fine.

closed 2023-07-01 by bobjacobsen