JMRI/JMRI #12134: Analog fast clock could use an optional AM/PM indicator

devel-bobm, 2023-06-28:
Some users may find that the Analog Fast Clock can use an indicator which tracks the fast-clock's AM/PM indication.

Indicators might be:
- A simple LED which lights when the clock is in the PM;
- A simple PM indicator;
- An indicator os AM or PM;
- some other choice;
- No AM/PM indication.

(My club's non-JMRI layout adds an LED on the analog mechanism.  I can imagime that an JMRI Analog Fast Clock may need something similar.)

PUlvestad, 2023-06-29:
The analog clock, when I open it shows AM or PM plus the clock rate right in the middle.
[image]

devel-bobm, 2023-08-09:
The time of day seems to "block out" the AM/PM and rate info at some times of day.  Especially when I looked at the icon!

But I cannot remove the Rate and/or AM/PM/Noon(!)/Midnight(!) indications.  

And it cannot display the rate if it is not integral, even though the Tools->Clocks->Set up Fast Clock tool can use up to three digits to the right of the decimal point.  Is there any real reason to support only integral rate display, even though the fast clock can (in some solutions) support non-integral rates?  If the reason is that the display cannot display enough characters, then why not?  And why cannot the display show more characters?

(I rarely find that an integral rate works for me or for the layouts that I operate on.  1.72:1, 1.5:1, 1.8:1 and 2.2:1 are ones that I have worked with around these parts...)

danielb987, 2023-08-10:
> The time of day seems to "block out" the AM/PM and rate info at some times of day. Especially when I looked at the icon!

Do you have the OS scale the screen? That could be the cause.

bobjacobsen, 2023-08-14:
I can't recreate the problem with the analog clock now showing non-integral rates.  The code explicitly supports up to 3 digits after the decimal point, ant that's what I see:

[image]

PR #12276 fixed a problem where the rate on the clock was not updating sometimes, perhaps that's what you were seeing.

devel-bobm, 2023-08-14:
This addressed the "tools->clock->analog clock" in `jmri.jmrit.display.analogclock.AnalogClockFrame`.  But it does not address the separately-implemented analog clock that can be added to a "panel", which is implemented via `jmri.jmrit.display.analogclock.AnalogClockFrame.java`.  That one is limited to integral rate display, even though the actual fast clock `Timebase` rate is not limited to an integer.

bobjacobsen, 2023-08-14:
Thanks @devel-bobm Have now added the code for that clock to PR 12276.

closed 2024-04-29 by github-actions[bot]