JMRI/JMRI #12148: Zimo MS+MN-Decoders CV115 is missing in "Shunt/Uncouple"

FancyRedCat, 2023-07-02:
I tried to to setup the uncouple function for a Zimo MS480 and MN300 decoder. For both decoders I cannot not setup CV115 in the "Shunt/Uncouple"-pane. I have to go to the "raw" CVs.

Maybe it it is only a display issue because the explanation for CV115 and CV116 are shown and the input field for CV116 doesn't really align with the text.

[image]

bobjacobsen, 2023-07-02:
Could you provide the value you’re seeing in CV65 please? 

DecoderPro is using that to decide whether to display the CV115 value or not.  It’s expecting it to be present only when the CV65 value is less than or equal to five.  Perhaps that’s wrong.

Bob

> I tried to to setup the uncouple function for a Zimo MS480 and MN300 decoder. For both decoders I cannot not setup CV115 in the "Shunt/Uncouple"-pane. I have to go to the "raw" CVs.
> Maybe it it is [...]

—
Bob Jacobsen
***@***.***

FancyRedCat, 2023-07-02:
In CV65 is in both decoders 229.
(CV7 is 4)

bobjacobsen, 2023-07-02:
Thank you!

DecoderPro is trying to decide whether to allow you to enter a 100’s digit (000, 100, 200) in that field, but it’s getting that wrong and omitting the entire CV.  

Do you know if the 100’s digit in the MS480 and MN300 decoders use that hundred’s digit for configuration?  

The label for that is "Hundreds digit defines waiting time between opening the coupling and driving away (0=0,3s / 1=2,5s / 2=1,0s)"

Do you know where I can find a manual for the MS480 and MN300 decoders?

Thank you again.

FancyRedCat, 2023-07-02:
In the Zimo manual for MS-docoders it is chapter 3.23 on page 40:
http://www.zimo.at/web2010/documents/MS-Sound-Decoders_EN.pdf

In German I found a newer manual for MS and MN-Decoders: http://www.zimo.at/web2010/documents/MS-MN-Decoder.pdf (chapter 3.23 on page 42). In this document it seems to be the same configuration for both decoder families.
So I think you can use for both decoder families the english MS-manual for the uncoupling CVs. (I setup the same values for both locos.)

And thank you for your fast response.

bobjacobsen, 2023-07-02:
PR #12149 has a fix for this.  Will let you know when it's in a development / nightly build so you can try it out.

bobjacobsen, 2023-07-03:
The PR has been merged.  Installers with the fix are available at

https://builds.jmri.org/jenkins/job/development/job/packages/

The fix will be in the next test release in about 10 days.

closed 2023-07-03 by bobjacobsen