JMRI/JMRI #12150: Layout Editor Shapes cause Java Exception

geowar1, 2023-07-03:
From: [Phil in gorgeous Young Harris](mailto:philabernathy@yahoo.com?subject=Re:%20Layout%20Editor%20Shapes%20cause%20Java%20Exception)
Date: Sun, 02 Jul 2023 06:41:27 PDT 
Howdy all.  Windows 10, Java 11.0.16, JMRI 5.4, LocoNet Simulator.

When I right-click on my "Shape" in Layout Editor, it causes a Java error.  I added a shape, and extended it into a straight line.  However, if I right-click on it, I get a Java exception error.  I made a small Panel to test to see whether I had caused the issue with other things, but I get the error in the new panel too.  Here is the System Console for that little test Panel: 

2023-07-02 09:11:57,918 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2023-07-02 09:11:57,921 util.Log4JUtil                        INFO  - This log is appended to file: C:\Users\Phil\JMRI\log\messages.log [main]
2023-07-02 09:11:57,921 util.Log4JUtil                        INFO  - This log is stored in file: C:\Users\Phil\JMRI\log\session.log [main]
2023-07-02 09:11:57,951 apps.Apps                             INFO  - PanelPro version 5.4+R9005ecb4ac starts under Java 11.0.16 on Windows 10 amd64 v10.0 at Sun Jul 02 09:11:57 EDT 2023 [main]
2023-07-02 09:12:03,001 apps.Apps                             INFO  - Starting with profile Loconet_Simulator.3ed9fa54 [main]
2023-07-02 09:12:03,414 node.NodeIdentity                     INFO  - Using jmri-fmtmabGXbhNiaatlSy0sD6-3ed9fa54 as the JMRI storage identity for profile id 3ed9fa54 [AWT-EventQueue-0]
2023-07-02 09:12:04,645 loconet.LnPacketizer                  INFO  - lnPacketizer Started [main]
2023-07-02 09:12:04,664 hexfile.LnHexFilePort                 INFO  - LocoNet Simulator Started [LocoNet HexFileFrame]
2023-07-02 09:12:05,132 roster.Roster                         INFO  - Reading roster file with rootFromName(C:\Download\Model Railroad\roster.xml) [main]
2023-07-02 09:12:05,203 util.FileUtilSupport                  INFO  - File path program: is C:\Program Files (x86)\JMRI\ [main]
2023-07-02 09:12:05,204 util.FileUtilSupport                  INFO  - File path preference: is C:\Download\Model Railroad\ [main]
2023-07-02 09:12:05,204 util.FileUtilSupport                  INFO  - File path profile: is C:\Users\Phil\JMRI\Loconet_Simulator\ [main]
2023-07-02 09:12:05,205 util.FileUtilSupport                  INFO  - File path settings: is C:\Users\Phil\JMRI\ [main]
2023-07-02 09:12:05,205 util.FileUtilSupport                  INFO  - File path home: is C:\Users\Phil\ [main]
2023-07-02 09:12:05,206 util.FileUtilSupport                  INFO  - File path scripts: is C:\Download\Model Railroad\Jython\ [main]
2023-07-02 09:12:06,731 PanelPro.PanelPro                     INFO  - Main initialization done [main]
2023-07-02 09:12:21,040 audio.JoalAudioFactory                INFO  - Initialised JOAL using OpenAL: vendor - Creative Labs Inc. version - 1.1 [Listed Table Generation]
2023-07-02 09:20:59,281 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
java.lang.IllegalArgumentException: SHAPE_CENTERis not a valid SHAPE_POINT
at jmri.jmrit.display.layoutEditor.HitPointType.shapePointIndex(HitPointType.java:354)
at jmri.jmrit.display.layoutEditor.LayoutShape.showShapePopUp(LayoutShape.java:427)
at jmri.jmrit.display.layoutEditor.LayoutEditor.showEditPopUps(LayoutEditor.java:3813)
at jmri.jmrit.display.layoutEditor.LayoutEditor.mouseReleased(LayoutEditor.java:3583)
at jmri.util.swing.JmriMouseListener$1.mouseReleased(JmriMouseListener.java:33)
at java.desktop/java.awt.Component.processMouseEvent(Unknown Source)
at java.desktop/javax.swing.JComponent.processMouseEvent(Unknown Source)
at java.desktop/java.awt.Component.processEvent(Unknown Source)
at java.desktop/java.awt.Container.processEvent(Unknown Source)
at java.desktop/java.awt.Component.dispatchEventImpl(Unknown Source)
at java.desktop/java.awt.Container.dispatchEventImpl(Unknown Source)
at java.desktop/java.awt.Component.dispatchEvent(Unknown Source)
at java.desktop/java.awt.LightweightDispatcher.retargetMouseEvent(Unknown Source)
at java.desktop/java.awt.LightweightDispatcher.processMouseEvent(Unknown Source)
at java.desktop/java.awt.LightweightDispatcher.dispatchEvent(Unknown Source)
at java.desktop/java.awt.Container.dispatchEventImpl(Unknown Source)
at java.desktop/java.awt.Window.dispatchEventImpl(Unknown Source)
at java.desktop/java.awt.Component.dispatchEvent(Unknown Source)
at java.desktop/java.awt.EventQueue.dispatchEventImpl(Unknown Source)
at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
at java.base/java.security.AccessController.doPrivileged(Native Method)
at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
at java.desktop/java.awt.EventQueue$5.run(Unknown Source)
at java.desktop/java.awt.EventQueue$5.run(Unknown Source)
at java.base/java.security.AccessController.doPrivileged(Native Method)
at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
at java.desktop/java.awt.EventQueue.dispatchEvent(Unknown Source)
at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
at java.desktop/java.awt.EventDispatchThread.run(Unknown Source)
 

I was surprised when I saw "Shapes" as a menu option while editing on my Layout editor, so I thought that I would draw a box around a section of the Layout Editor Panel.  I got a box by adding four shapes and extending them into lines (right-click and drag), but always got the Java error when right-clicking on the center circle of the "Shape".

Phil

geowar1, 2023-07-03:
See PR #12151

danielb987, 2023-07-05:
PR #12151 is merged. Is this issue completed?

geowar1, 2023-07-06:
IMHO… yes… Sent from GeoWar’s iPad Pro

closed 2023-07-06 by danielb987