JMRI/JMRI #12177: Feature Request: Reporter appearance on Layout Editor Panel

steveasm, 2023-07-12:
Currently, if a reporter is inserted into a Panel created with Layout Editor it appears as <no report> or, if available, the value of the reporter. I have many reporters on my Layout Editor Panel and it gets very busy with the <no report> s everywhere, with some overlapping others.  It would be good if one could either choose their own label for the reporters and/or have them be blank unless there is a current value for the reporter.

bobjacobsen, 2023-07-12:
(Edited the lead comment to make the `<no report>` visible.

bobjacobsen, 2023-07-12:
It looks like the `<no report>` comes from java/src/jmri/jmrit/display/DisplayBundle.properties so it would be possible to replace that.  (There's also a definition under java/src/jmri/server/json that seems unrelated to this request)

bobjacobsen, 2023-07-12:
See [JMRIusers thread](https://groups.io/g/jmriusers/topic/100102025#219149)

bobjacobsen, 2023-07-13:
@steveasm One suggestion on the [JMRIusers thread](https://groups.io/g/jmriusers/topic/100102025#219149) was to display a hyphen ("-") when there's no report available.  Would that work?

bobjacobsen, 2023-07-14:
#12181 closes this by displaying a hypen

closed 2023-07-14 by bobjacobsen