JMRI/JMRI #12180: Hornby HM700 Decoder Support

ShawGawby, 2023-07-13:
Is it possible to add support for the new Hornby HM7000 decoders in Decoder pro / JMRI? 

From Hornby's documentation, it appears the decoder has to be set up via their bluetooth based app, which once configured,  enables the decoder to managed / controlled via their proprietary bluetooth app, or handed off to a standard DCC controller.

It would be great if the decoder could be read and the loco added to the Decoder Pro roster.

bobjacobsen, 2023-07-13:
Could you provide a link to that documentation? Thanks

ShawGawby, 2023-09-18:
Hi Bob, 

Here is the link to the quick start guide:

https://d63oxfkn1m8sf.cloudfront.net/2616/8717/6617/HM7000-TXS_-_Quick_Start_Guide_EN-FR.pdf

Here is the link to the full guide.

https://d63oxfkn1m8sf.cloudfront.net/8016/8717/6643/HM7000_Series_Decoders_Reference_Manual.pdf

Andrew

bobjacobsen, 2023-09-18:
Thanks for the links.  The full guide is really complete.

Do you have one of these decoders?  If so, could you please read CVs 7, 8, 47, 48, 49 and 159 and let me know what you find?  Thanks in advance.

ShawGawby, 2023-09-19:
Hi Bob,

Please find details of the CV's values as requested:

CV7: 254
CV8: 48
CV47: 0
CV48:0
CV49: 5
CV159 (Volume F0): 40 

For information, the above was obtained using the Hornby HM7000 app on an iPad, using the Bluetooth control. The decoder is an HM7000-21TXS. 

When you query the CV values of the locomotive via the app, if the locomotive decoder is set to DCC control as opposed to Bluetooth, the app takes back control of the locomotive using Bluetooth.  I suppose this makes sense. 

The other thing I noticed is the app tells you if the CV values aren't in sync.  From what I've read before, I'm assuming this is against the info stored for the Bluetooth decoder on the user's Hornby cloud account.   

Let me know if you need anything else. 

Andrew

bobjacobsen, 2023-09-19:
That's a bit unfortunate.

JMRI uses CV7 to make a first estimate of the decoder ID, then goes to specific CVs for more detail.  Previous Hornby decoders have used CV159 for this extra information.  The HM7000 uses CV159 for sound, and other CVs for ID.

Normally, this is not a problem:  That first estimate with CV7 can be used to tell which CVs to look at next.  Unfortunately, we already have Hornby decoders that use CV7=48.  We won't be able to tell those apart at that initial step!

Not sure what to do about this.  It might be necessary to have the HM7000 only be manually selectable, without JMRI being able to automatically identify it.

Working  on it (but slowly)

bobjacobsen, 2023-09-19:
When you get a chance, could you read CV47, CV48 and CV49?

Also, which kind of decoder do you have?  E.g. Diesel, Steam, Electrical, etc

Thanks!

ShawGawby, 2023-09-22:
Hi Bob,

CV47 = 0
CV48 = 0
CV49 = 5

The class 66 is a diesel loco.

Let me know if you need anything else.  I do have a spare Hornby HM70000 decoder and it's pretty easy to create another profiles of a steam loco etc. 

Andrew

bobjacobsen, 2023-09-23:
PR #12441 is the first part for this.  There's still more to do on a decoder definition, but at least #12441 gets the code updates in.

ShawGawby, 2024-04-17:
Hi Bob,

In trust you are well? I was wondering if there is any news on the Hornby HM7000 decoder support yet and if you need any further information? Andy

PUlvestad, 2025-01-02:
#13727

closed 2026-10-06 by bobjacobsen