JMRI/JMRI #12190: Missing ICONs after upgrade to 5.4

SJROW2, 2023-07-17:
[session.log](https://github.com/JMRI/JMRI/files/12070782/session.log)
[systemInfo.txt](https://github.com/JMRI/JMRI/files/12070784/systemInfo.txt)

Issue Type: __Bug__

This is the 2nd Linux-Mint System that after upgrading the signal head icons came up with the Red-X and the block sensor icons are missing.

I'm using Dropbox as the JMRI directories.  The executables are on the local drive.

If I do a "build signal heads" the Red-Xs are replaced with the proper icons.

PanelPro 5.4+R9005ecb4ac
OpenJDK 64-Bit Server VM Ubuntu 11.0.19+7-post-Ubuntu-0ubuntu122.04.1 mixed mode, sharing
Linux 5.15.0-75-generic amd64

SJROW2, 2023-07-17:
> Based on the system log, the previous JMRI install was located at ~/.jmri. This resulted in the icon references using the "settings:" keyword instead of the "program:" keyword.
>
> Replacing "settings:" with "program:" in the xml file should take care of the issue.

That worked. Thanks.

Get BlueMail for Android<https://bluemail.me>

danielb987, 2023-07-17:
@SJROW2 
Welcome to the JMRI project!

Is the problem solved? Can this issue be closed?

SJROW2, 2023-07-18:
Yes.  JMRI support is amazing.

Get BlueMail for Android<https://bluemail.me>

closed 2023-07-18 by danielb987