JMRI/JMRI #12193: TAMS_LD-G-9.xml file

warnerthuis, 2023-07-18:
Hello,
I created and tested the xml file for the Tams LD-G-9 decoder.

-Warner-

[TAMS_LD-G-9.zip](https://github.com/JMRI/JMRI/files/12088974/TAMS_LD-G-9.zip)

warnerthuis, 2023-07-19:
I found an error in the file. In line 78 there is "min=23 max=0" And that should be the other way around of course.
The thing is I copied that piece of code from a file and I cannot find that file anymore.
I include the corrected one.
[TAMS_LD-G-9.zip](https://github.com/JMRI/JMRI/files/12098827/TAMS_LD-G-9.zip)

bobjacobsen, 2023-07-19:
The max="0" appears in two files:
```
% grep -r 'max="0"' xml 
xml/decoders/TAMS_LD_30.xml:        <decVal max="0" min="255"/>
xml/decoders/TAMS_LD_30.xml:        <decVal max="0" min="255"/>
xml/decoders/TAMS_LD_30.xml:        <decVal max="0" min="255"/>
xml/decoders/TAMS_LD_30.xml:        <decVal max="0" min="255"/>
xml/decoders/TAMS_LD_30.xml:        <decVal max="0" min="255"/>
xml/decoders/Fleischmann_69687401.xml:        <decVal max="0" min="255"/>
xml/decoders/Fleischmann_69687401.xml:        <decVal max="0" min="255"/>
```

I didn't find a min="23"

bobjacobsen, 2023-07-26:
See PR #12220 which will merge the 2nd version of the definition. Thanks!

bobjacobsen, 2023-07-29:
Closed by merging #12220

closed 2023-07-29 by bobjacobsen