JMRI/JMRI #12204: C/MRI's serial.SerialNode.createOutPacket array overrun error?

devel-bobm, 2023-07-21:
I have seen an array overrun today in `serial.SerialNode.createOutPacket()`.  I have only noticed this today.  

Have I found a real bug here?  Is there supposed to be some external mechanism that prevents the outputs from changing when `serial.SerialNode.createOutPacket()` runs?

Here's what I know:

I am getting an occasional "array overrun" problem under C/MRI Simulation which is caused when the outputs seem to have CHANGED between the time of calculation of the "DLE bytes" and the time the data and any DLE bytes are filled into the array.  The byte length is allocated to match the DLE count.  But if the data changes such that the number of DLEs _increases_ between the original computation and the number of DLEs when the data/DLEs are added to the byte array, the byte array "over-fills" and the code generates an error.  

I believe the problem occurs with real serial ports, too.  I am too-far-away from the C/MRI to test any real C/MRI equipment...

Typically, this problem would be solved by "cloning" the output array, and working with the clone for both computing the # of DLEs and inserting the data and the required DLEs, before sending the serial data.

I figure to "clone" the array and use the clone for calculating the DLEs and copying the cloned data to the output data, with the required DLEs.  This is shown below:

```
    public AbstractMRMessage createOutPacket() {
        // Count the number of DLE's to be inserted
        int nOutBytes = numOutputCards() * (bitsPerCard / 8);
        int nDLE = 0;
        byte[] oA = new byte[256]; // current values of the output bits for this node
        
        oA = outputArray.clone();

        for (int i = 0; i < nOutBytes; i++) {
            if ((oA[i] == 2) || (oA[i] == 3) || (oA[i] == 16)) {
                nDLE++;
            }
        }
        // Create a Serial message and add initial bytes
        SerialMessage m = new SerialMessage(nOutBytes + nDLE + 2);
        m.setElement(0, getNodeAddress() + 65); // node address
        m.setElement(1, 84);             // 'T'
        // Add output bytes
        int k = 2;
        for (int i = 0; i < nOutBytes; i++) {
            // perform C/MRI required DLE processing
            if ((oA[i] == 2) || (oA[i] == 3) || (oA[i] == 16)) {
                m.setElement(k, 16);  // DLE
                k++;
            }
            // add output byte
            m.setElement(k, oA[i]);
            k++;
        }
        return m;
    }
```
Is this reasonable and necessary?  Or am I barking up the wrong tree?

bobjacobsen, 2023-07-23:
This looks like a good fix to me.

Only thing I'd suggest is that I think you don't have to allocate an array before you load it from the clone operation.  So those two lines could be

```
        byte[] oA = outputArray.clone(); // current values of the output bits for this node
```

which saves the extra allocation of the `new byte[256]`.  Not a big thing though.

devel-bobm, 2023-07-23:
I've been wondering: how does JMRI have this problem?  I __thought__ that there would be one processor running the JMRI process, so I thought it would not have this problem.  I have certainly not noticed the problem before.

Then I thought that the "Jython" process could be run as a separate process, and I did have a jython process running.  Could this be the source of such an independant process?  If true, it could be a problem for any system connection which needs to be able to add "padding" data because of byte-specific data...  This would need further investigation by those who are knowledgable!

@bobjacobsen , thanks for the hint/correction!  I plan to submit the "fix".  But I certainly do not understand how I could possibly "test" this.

bobjacobsen, 2023-07-24:
Were you seeing it often, or just a few times?

JMRI is multi-threaded.  There's a primary thread that handles all the GUI stuff, signaling logic, etc.  Then there are (usually) separate read and write threads for each system connection.  That's done so that input and output characters can be processed promptly.  So it's possible that the main thread will change something in the middle of the output thread processing for DLEs.  It's unlikely, as it requires a pretty specific lineup, and the time that the output thread is doing this processing is quite small.  But it certainly can happen!

I'll take a look around other connections to see if I can find similar code.

Thanks for tracking this down.

PR #12228 (2023-07-29): C/MRI serial output ignores new outputs if currently making a buffer
C/MRI could generate new output while an "T"ransmit buffer was being generated.  This could cause the data to change _after_ the "DLE" bytes were pre-calculated but before the data was put into the buffer.  This can be a problem if the number of actual DLE bytes is greater than the calculated number - causing a "buffer out of range" error.

This code grabs a clone of the output data for the node, and uses that clone to pre-compute the DLE bytes and data bytes.  Then it uses the clone data to build the actual "T"ransmit messag [... cut]
closed 2023-07-29 by devel-bobm