JMRI/JMRI #12207: DCC++ Ethernet Fails to Restart Status Polling on Reconnect

DeanCording, 2023-07-23:
The DCC++ Ethernet connection uses a TCP network connection to talk with the DCC++ base station.  The connector regularly polls the status of base station by sending a <s> request.

The TCP connection can be lost if the base station is rebooted, or there is an interruption of the physical network connection.  The DCC++ Ethernet connector correctly identifies and reports the loss of connection.  It subsequently attempts to reestablish this connection, which it successfully does so when the issue is resolved. 

However, it fails to restart the regular polling of the status of the DCC++ base station.  As a result, the status of turnouts and sensors may be incorrect in JMRI as their states will not be updated until they next change, which may be never.

mstevetodd, 2023-07-24:
@DeanCording It would be very helpful if you could attach a copy of your "session.log" file following a disconnection and subsequent reconnection. It might show messages associated with this failure to restart the polling.

DeanCording, 2023-07-25:
This shows the disconnect and reconnection:

2023-07-25 17:18:27,993 jmrix.AbstractMRTrafficController     ERROR - run: Exception: java.net.SocketException: Connection reset in class jmri.jmrix.dccpp.network.DCCppEthernetPacketizer [network.DCCppEthernetPacketizer Receive thread]
java.net.SocketException: Connection reset
	at java.base/sun.nio.ch.NioSocketImpl.implRead(NioSocketImpl.java:323)
	at java.base/sun.nio.ch.NioSocketImpl.read(NioSocketImpl.java:350)
	at java.base/sun.nio.ch.NioSocketImpl$1.read(NioSocketImpl.java:803)
	at java.base/java.net.Socket$SocketInputStream.read(Socket.java:966)
	at java.base/java.io.DataInputStream.read(DataInputStream.java:151)
	at jmri.jmrix.AbstractMRTrafficController.readByteProtected(AbstractMRTrafficController.java:959)
	at jmri.jmrix.dccpp.DCCppPacketizer.loadChars(DCCppPacketizer.java:149)
	at jmri.jmrix.AbstractMRTrafficController.handleOneIncomingReply(AbstractMRTrafficController.java:1063)
	at jmri.jmrix.AbstractMRTrafficController.receiveLoop(AbstractMRTrafficController.java:880)
	at jmri.jmrix.AbstractMRTrafficController$2.run(AbstractMRTrafficController.java:803)
	at java.base/java.lang.Thread.run(Thread.java:833)
2023-07-25 17:18:27,995 jmrix.AbstractMRTrafficController     WARN  - Exception java net:  [network.DCCppEthernetPacketizer Receive thread]
java.net.SocketException: Connection reset
	at java.base/sun.nio.ch.NioSocketImpl.implRead(NioSocketImpl.java:323)
	at java.base/sun.nio.ch.NioSocketImpl.read(NioSocketImpl.java:350)
	at java.base/sun.nio.ch.NioSocketImpl$1.read(NioSocketImpl.java:803)
	at java.base/java.net.Socket$SocketInputStream.read(Socket.java:966)
	at java.base/java.io.DataInputStream.read(DataInputStream.java:151)
	at jmri.jmrix.AbstractMRTrafficController.readByteProtected(AbstractMRTrafficController.java:959)
	at jmri.jmrix.dccpp.DCCppPacketizer.loadChars(DCCppPacketizer.java:149)
	at jmri.jmrix.AbstractMRTrafficController.handleOneIncomingReply(AbstractMRTrafficController.java:1063)
	at jmri.jmrix.AbstractMRTrafficController.receiveLoop(AbstractMRTrafficController.java:880)
	at jmri.jmrix.AbstractMRTrafficController$2.run(AbstractMRTrafficController.java:803)
	at java.base/java.lang.Thread.run(Thread.java:833)
2023-07-25 17:18:27,996 jmrix.AbstractMRTrafficController     ERROR - Exit from rcv loop in class jmri.jmrix.dccpp.network.DCCppEthernetPacketizer [network.DCCppEthernetPacketizer Receive thread]
2023-07-25 17:18:28,998 util.TimerUtil                        WARN  - During schedule() [Connection Recovery 192.168.1.26:2560]
java.lang.IllegalStateException: Task already scheduled or cancelled
	at java.base/java.util.Timer.sched(Timer.java:413)
	at java.base/java.util.Timer.schedule(Timer.java:260)
	at jmri.util.TimerUtil.schedule(TimerUtil.java:71)
	at jmri.jmrix.dccpp.network.DCCppEthernetAdapter.keepAliveTimer(DCCppEthernetAdapter.java:100)
	at jmri.jmrix.dccpp.network.DCCppEthernetAdapter.connect(DCCppEthernetAdapter.java:46)
	at jmri.jmrix.AbstractNetworkPortController.reconnectFromLoop(AbstractNetworkPortController.java:305)
	at jmri.jmrix.AbstractPortController$ReconnectWait.run(AbstractPortController.java:458)
	at java.base/java.lang.Thread.run(Thread.java:833)
2023-07-25 17:18:28,998 jmrix.AbstractPortController          INFO  - Reconnected to 192.168.1.26:2560 [Connection Recovery 192.168.1.26:2560]

DeanCording, 2023-07-25:
After a 14 minute delay, I get this message repeating:

2023-07-25 17:32:11,895 jmrix.AbstractMRTrafficController     ERROR - left timeout in unexpected state: 50 [network.DCCppEthernetPacketizer Transmit thread]
2023-07-25 17:32:12,095 jmrix.AbstractMRTrafficController     ERROR - left timeout in unexpected state: 50 [network.DCCppEthernetPacketizer Transmit thread]
2023-07-25 17:32:22,001 jmrix.AbstractMRTrafficController     ERROR - left timeout in unexpected state: 50 [network.DCCppEthernetPacketizer Transmit thread]
2023-07-25 17:32:22,402 jmrix.AbstractMRTrafficController     ERROR - left timeout in unexpected state: 50 [network.DCCppEthernetPacketizer Transmit thread]
2023-07-25 17:32:22,602 jmrix.AbstractMRTrafficController     ERROR - left timeout in unexpected state: 50 [network.DCCppEthernetPacketizer Transmit thread]
2023-07-25 17:32:22,702 jmrix.AbstractMRTrafficController     ERROR - left timeout in unexpected state: 50 [network.DCCppEthernetPacketizer Transmit thread]
2023-07-25 17:32:22,902 jmrix.AbstractMRTrafficController     ERROR - left timeout in unexpected state: 50 [network.DCCppEthernetPacketizer Transmit thread]
2023-07-25 17:32:23,203 jmrix.AbstractMRTrafficController     ERROR - left timeout in unexpected state: 50 [network.DCCppEthernetPacketizer Transmit thread]
2023-07-25 17:32:23,503 jmrix.AbstractMRTrafficController     ERROR - left timeout in unexpected state: 50 [network.DCCppEthernetPacketizer Transmit thread]

icklesteve, 2023-07-25:
For anyone looking at this, I'd be tempted to start by adding these methods to https://github.com/JMRI/JMRI/blob/master/java/src/jmri/jmrix/dccpp/network/DCCppEthernetAdapter.java .
Is the previous DCCppEthernetPacketizer being closed ok?  I've no hardware to test with.

```
    @Override
    protected void closeConnection(){
        log.info("Closing connection {}.",getCurrentPortName());
        if (keepAliveTimer != null ){
            keepAliveTimer.cancel();
        }
        super.closeConnection();
    }

    @Override
    protected void resetupConnection() {
        super.resetupConnection();
        keepAliveTimer();
    }
```

mstevetodd, 2023-08-01:
@icklesteve
<<Is the previous DCCppEthernetPacketizer being closed ok?>>
How can I determine this?

mstevetodd, 2023-08-02:
The problem seems to be that jmri.util.TimerUtil.schedule() will only start a task if its current state is VIRGIN (0). But the .cancel() method sets it to CANCELLED (3). So it will never restart.

mstevetodd, 2023-08-02:
@icklesteve If you'd like to give this a go, I have a DCC-EX on internet connection you can use for testing. Contact me via email.

mstevetodd, 2023-08-02:
Upon further reflection, the current code doesn't try to restart the keepAlive until the reconnection, but the task is still running, so no need to restart it!
I simply removed the cancel and attempt to restart it, and it is working fine for me. 
I'll create a PR with this and see what others think.

commit 407aaead (2023-08-03): Merge pull request #12248 from mstevetodd/master

Don't kill keepAlive on reconnection, closes #12207
commit 57a6ebc0 (2023-08-03): Don't kill keepAlive on reconnection, closes #12207
closed 2023-08-03 by danielb987