JMRI/JMRI #12208: Sprog 3 Command Station won't start with track power on

PUlvestad, 2023-07-23:
5.5.2plus, preferences are set for "Track Power at Startup" as ON.

PUlvestad, 2023-07-23:
Sprog 3 v4.3

crosland, 2023-07-24:
Hi Peter, It works OK for me in 5.4. I am not aware of any recent changes in this area.

PUlvestad, 2023-07-24:
I back dated to 5.0 and have the same issue

PUlvestad, 2023-07-24:
I switch over to Windows 10 which is running 5.4 and it displayed the same issue.

PUlvestad, 2023-07-24:
To clarify, on startup the power indication displays ON but there is no track power until the power button is cycled OFF then back ON.

PUlvestad, 2023-07-24:
I just did a voltage check (nothing accurate) but it appears there is track voltage at startup but it's not receiving a DCC signal until after a power cycle. Also, not sure if it's related but the USB light is on at startup but goes out after the power cycle.

crosland, 2023-07-25:
OK, I see it now. I was relying on my sound loco powering up, but it's set to do that as soon as it sees any valid DCC signal, even just preambles.

closed 2023-07-27 by PUlvestad