JMRI/JMRI #12226: Block Contents

tmf28, 2023-07-27:
[image]
[image]

Issue Type: __Feature Request__

To make Panel Pro more prototypical, an option to only show empty blocks in editor mode would be really nice.

The trick to hide the block by making it transparent or the same color as the panel background color takes away from being able to style the train ID in a prototypical maner.

With the AMTEC CTC system that we use at Amtrak on the NEC, train IDs are black text with a yellow background. So, when I configured Panel Pro like that, I have a bunch of small yellow squares on the track model.

 
PanelPro 5.5.2plus+jenkins+20230726T2155Z+R30ff23c880d
OpenJDK 64-Bit Server VM Azul Systems, Inc. 11.0.19+7-LTS mixed mode
Windows 11 10.0 amd64

dsand47, 2023-07-27:
Subject to further review, this looks feasible.

An alternate approach is to use LogixNG:  https://www.jmri.org/help/en/html/tools/logixng/reference/chapter5.shtml#digital_action_display

dsand47, 2023-07-27:
Here is a LogixNG solution.
```
       1:  LogixNG: IQ:AUTO:0001
       2:      ConditionalNG: IQC:AUTO:0001
       3:          ! A
       4:              Many
       5:                 ::: Local variable "block", init to None ""
       6:                 ::: Local variable "value", init to None ""
       7:                 ::: Local variable "id", init to String ""
       8:                  ! A1
       9:                      // Create a list of blocks whose block content labels
      10:                      // will be shown or hidden based on the block value.
      11:                      Listen on beans
      12:                  ! A2
      13:                      If Then Else. Always execute
      14:                          ? If
      15:                              Local variable block is not null
      16:                          ! Then
      17:                              Many
      18:                                  ! A1
      19:                                      // The id of a block conent label is the same as the user name for the block.
      20:                                      Set variable id to the value of variable block
      21:                                  ! A2
      22:                                      If Then Else. Always execute :: Evaluate needed expressons
      23:                                          ? If
      24:                                              Digital Formula: E1 || E2
      25:                                                  ?* E1
      26:                                                      Local variable value is null
      27:                                                  ?* E2
      28:                                                      Local variable value is equal to ""
      29:                                          ! Then
      30:                                              Many
      31:                                                  ! A1
      32:                                                      Log data: Only text: Hide the block
      33:                                                  ! A2
      34:                                                      Set icon/label "by local variable "id"" on panel "My Layout" to "Hide"
      35:                                          ! Else
      36:                                              Many
      37:                                                  ! A1
      38:                                                      Log data: Only text: Show the block
      39:                                                  ! A2
      40:                                                      Set icon/label "by local variable "id"" on panel "My Layout" to "Show"
      41:                          ! Else
      42:                              
```

tmf28, 2023-07-27:
Thanks, I'll try it out when I get home tonight.

tmf28, 2023-07-28:
I created this but it doesn't seem to be working.  I'm sure I am missing something and/or doing it wrong.

[image]

tmf28, 2023-07-28:
I fiddled with this again this morning.  Still not able to get it to work.  I'll wait to see if an option to only show empty blocks in editor mode is added.

dsand47, 2023-07-28:
Make sure that you have added the blocks in the Listen on Beans dialog.

[image]

tmf28, 2023-07-28:
Am I supposed to be using the block system names? 

[image]

I also noticed that is am missing this part.  Not sure how to add that.  It is selected though.

[image]
[image]

tmf28, 2023-07-28:
[image]
[image]
[image]
[image]

tmf28, 2023-07-28:
OK, I figured out what you meant by adding the blocks to the bean.

[image]

Let me test it again.

tmf28, 2023-07-28:
Still showing the empty blocks.

[image]

dsand47, 2023-07-28:
I see that the help content is not complete for the set icon action.  

You need to assign an id to each icon, in this case block content labels.
[image]

For the sample LogixNG, the id has to match the block user name.  This provides the link between the block object and the block contents label.
[image]

tmf28, 2023-07-28:
OK, I gave each block an id that matched the block label.  On my test setup, I just have 3 blocks.  Labeled 4, 5 and 6.

Saved, re-launched JMRI.  Still have the empty yellow boxes. I feel we are getting close though! 

[image]
[image]
[image]

dsand47, 2023-07-28:
Did you add and remove a value for 4 and 5?

Use the "Hide when not editing" option to insure an hidden label during file loading if there is no block value.  LogixNG will then take care of show/hide based on subsequent changes.

tmf28, 2023-07-28:
> Did you add and remove a value for 4 and 5?

If you mean moving the loco in and out of those blocks, yes.

> Use the "Hide when not editing" option to insure a hidden label during file loading if there is no block value. LogixNG will then take care of show/hide based on subsequent changes.

I did not have that option checked.  I'll have to test again when I get home tonight. Will report back then.  Thanks.

tmf28, 2023-07-29:
It works until a train enters and exits a block, then it leaves the empty yellow box again.  I went back to edit mode and the "Hide when not editing" option was no longer checked.

tmf28, 2023-07-29:
1:  LogixNG: IQ:AUTO:0001
       2:      ConditionalNG: IQC:AUTO:0001
       3:          ! A
       4:              Many
       5:                 ::: Local variable "block", init to None ""
       6:                 ::: Local variable "value", init to None ""
       7:                 ::: Local variable "id", init to String ""
       8:                  ! A1
       9:                      Listen on beans
      10:                  ! A2
      11:                      If Then Else. Always execute
      12:                          ? If
      13:                              Local variable block is not null
      14:                          ! Then
      15:                              Many
      16:                                  ! A1
      17:                                      Set variable id to the value of variable block
      18:                                  ! A2
      19:                                      If Then Else. Always execute
      20:                                          ? If
      21:                                              Digital Formula: E1 || E2
      22:                                                  ?* E1
      23:                                                      Local variable value is null
      24:                                                  ?* E2
      25:                                                      Local variable value is equal to ""
      26:                                          ! Then
      27:                                              Many
      28:                                                  ! A1
      29:                                                      Log data: Only text: Hide the block
      30:                                                  ! A2
      31:                                                      Set icon/label "by local variable "id"" on panel "My Layout 2" to "Hide"
      32:                                          ! Else
      33:                                              Many
      34:                                                  ! A1
      35:                                                      Log data: Only text: Show the block
      36:                                                  ! A2
      37:                                                      Set icon/label "by local variable "id"" on panel "My Layout 2" to "Show"
      38:                          ! Else
      39:

dsand47, 2023-07-29:
The attached zip file contains a complete JMRI profile.  Place it next to your other profiles and you should be able to select it from the profile selector or by using Preferences => Config Profiles.

There were a number of issues to resolve.
- The "block value" event only occurs when the value is changed using the block table.  Block tracking does not create the "value" event when the values are moved between blocks.
- The "block status" event only shows whether the block is occupied or not occupied.
- LogixNG does not have direct access to the block value.  Common values such as system names, user names, etc., are available to LogixNG Formulas.

To resolve these issues, a custom LogixNG Function was implemented using Jython.  See the "Add new function with Jython" section at https://www.jmri.org/help/en/html/tools/logixng/reference/chapter9.shtml.  The Jython file is loaded before loading the xml data file.  The demo uses start up actions.

----

The current state of the "Hide when not editing" option is stored along with the rest of the tables and panels.  The state changes when LogixNG does a show or hide.

This has implications for block content labels.  When PanelPro is stopped, the current block values are written to the blockvalues.xml file.  If a block contents label is shown when a store is done, the option will be set inactive.  If the block is not occupied when the xml data file is loaded, the blockvalues.xml process will set the block empty but the block contents label option will still be inactive.  This results in the empty square.

----

[HideShowDemo.jmri.zip](https://github.com/JMRI/JMRI/files/12207919/HideShowDemo.jmri.zip)

tmf28, 2023-07-29:
I will give it a try when I get home tomorrow morning. Thank you!

tmf28, 2023-07-30:
Not having any luck with it.  Train IDs are not showing.

dsand47, 2023-07-30:
> Not having any luck with it. Train IDs are not showing.

Interesting.  I assume you are talking about the demo profile.

When it starts, the left block should show 1234.  As you click on the sensor icons, the value should move to adjacent blocks.

Are there any errors in the system console?

tmf28, 2023-07-30:
It does show 1234.  I didn't test it that way, so it is probably an error on my part.  I added a train to the roster and assigned my Z21 sensors and reporters to 3 of the 4 blocks. (my test track is only 3 blocks).  Then I ran a my loco.  I saw the occupancy but not the ID. 

I'm back at work again after a 16 hr shift.  I'm only working 8 hrs tonight.  I'll check for errors and give it another try when I get home tonight. Off tomorrow and Tuesday so I'll have plenty of time to work on this.   

I hope to get it like this without the empty yellow square above every unoccupied block. :)  Thanks again for your help.

[image]

tmf28, 2023-07-31:
I tested the demo and also did not get the 1234 ID moving.

[image]

dsand47, 2023-07-31:
Don't spend any more time on the demo.  

I have the code changes pretty much done.

tmf28, 2023-07-31:
OK, thank you!

Tommy

closed 2023-08-02 by bobjacobsen

tmf28, 2023-08-02:
Thank you guys!

bobjacobsen, 2023-08-02:
There are development installers that include this PR at https://builds.jmri.org/jenkins/job/development/job/packages/

tmf28, 2023-08-02:
I'll install it tonight when I get home.  I appreciate you guys adding this feature.  Thanks again.  

Tommy

tmf28, 2023-08-03:
I have the "Hide when empty and not editing" option checked for each block.  Is there a way to force Panel Pro to know which blocks are empty and should be hidden upon loading the panel?  The block contents boxes only become hidden if I occupied, then unoccupied the block. 

https://github.com/JMRI/JMRI/assets/139667932/afea2f31-b500-46a4-9d69-36206edd958f

dsand47, 2023-08-03:
Try setting both hide options.

https://github.com/JMRI/JMRI/assets/139667932/afea2f31-b500-46a4-9d69-36206edd958f

tmf28, 2023-08-03:
Yes, that did the trick!

Thanks,

Tommy

https://github.com/JMRI/JMRI/assets/139667932/afea2f31-b500-46a4-9d69-36206edd958f

tmf28, 2023-08-03:
https://github.com/JMRI/JMRI/assets/139667932/1c2af7fd-82ad-4d6c-a68a-4959012f8d23