JMRI/JMRI #12244: "Edit Current Memory Value" dialog should support pressing [Enter] to accept

mstevetodd, 2023-08-01:
Currently, editing a Memory Value on a Layout Editor panel requires that the user press the [OK] button using the mouse.
It would be faster if the user could simply press [Enter] to accept the new value.
Perhaps setting the [OK] button as default would work?

[image]

dsand47, 2023-08-01:
> I _think_ that the `options[2]` argument is putting the cursor into the entry field by default, which seems right to me. Not sure how to change the behavior of "return" in a `JOptionPane.showOptionDialog`.

The block contents icon uses options[1] which requires clicking in the text field to enter the data.

For the  memory input dialog, I use tab-space to trigger the OK button.

bobjacobsen, 2023-08-01:
I think we'd get the desired behavior if we change this to a `showInputDialog` _except_ that I can't see how to get the initial value into the string entry field.

danielb987, 2023-08-05:
Is someone working on this? A suggestion is to add a `showInputDialog` method that does what we want to some class in jmri.util.

bobjacobsen, 2023-08-05:
I was working on changing it to a `showInputDialog`, but I wasn't able to figure out how to get it to display the current value at first appearance.  Haven't worked on it since.  It would be great to have a common solution to that.

danielb987, 2023-08-05:
> I think we'd get the desired behavior if we change this to a `showInputDialog` _except_ that I can't see how to get the initial value into the string entry field.

Doesn't this method do the job? Or that is needed that this method cannot do?
https://docs.oracle.com/javase/8/docs/api/javax/swing/JOptionPane.html#showInputDialog-java.awt.Component-java.lang.Object-java.lang.Object-

```
public static String showInputDialog(Component parentComponent,
                                     Object message,
                                     Object initialSelectionValue)
```

If not, how should a new `showInputDialog` method look like? Which parameters should it take?

dsand47, 2023-08-05:
The memory label dialog has the focus in the text field which requires changing the focus to the OK button, which is normally done by doing a click or using tab.

The block label dialog has the focus on the OK button.  To change the block contents, the focus has to be moved to the text field.  After changing the content, the enter key can be used to trigger the OK or the OK button can be clicked.

The request is to have the "edit focus" in the text field and the "enter key focus" be the OK button.

bobjacobsen, 2023-08-05:
Thanks @danielb987, I had overlooked that one.

@dsand47 The nice thing about a native `showInputDialog` is that you get both the initial focus on the value _and_ having return accept the value without having to use the mouse.

closed 2023-08-07 by danielb987