JMRI/JMRI #12268: Incorrect error with blank ID tag?

jerryg2003, 2023-08-10:
Due to a error on my part, the incoming "msgValue" in this code fragment was blank:

tagID = idtags.getByTagID(msgValue)

There are several things I don't understand: The error showing up in the System Console below says that Reporter must start with "ID" but in fact it is IDTag that must start with "ID".  Since this was a "getByTagID" it shouldn't have needed the "ID" anyway.  And should it just have returned tagID = None, which was what I would have expected? [MemSensArray is the name of my script] 

2023-08-10 17:18:47,867 managers.AbstractManager              ERROR - Invalid system name for Reporter: System name must start with "ID". [MemSensArray]
2023-08-10 17:18:47,874 automat.AbstractAutomaton             WARN  - Unexpected Exception ends AbstractAutomaton thread [MemSensArray]
jmri.NamedBean$BadSystemNameException: System name must start with "ID".
 at jmri.Manager.makeSystemName(Manager.java:177)
 at jmri.managers.AbstractManager.makeSystemName(AbstractManager.java:98)
 at jmri.Manager.makeSystemName(Manager.java:150)
 at jmri.Manager.makeSystemName(Manager.java:128)
 at jmri.managers.AbstractProxyManager.makeSystemName(AbstractProxyManager.java:547)
 at jmri.managers.ProxyIdTagManager.getByTagID(ProxyIdTagManager.java:167)

bobjacobsen, 2023-08-10:
The message about needing an "ID" prefix is because it _is_ looking for a tag.  The start of the trace you show (at the bottom) your a call to getByTagID. So it's trying to make a valid tag ID from the contents of `msgValue` so it can use that to do the lookup.

danielb987, 2023-08-11:
@jerryg2003 
IDtags are NamedBeans with a system name and a user name. The method tries to convert the ID to a valid IDtag system name and then lookup the IDtag by this system name.

If the ID doesn't start with `ID`, it adds these letters to the front of the ID. And then lookup the name. But an empty ID results in the system name `ID`, which is not a valid system name.

bobjacobsen, 2023-08-14:
PR #12271 fixed the "Reporter" in the first line of the error message to now refer to ID tag.

closed 2023-08-14 by bobjacobsen