JMRI/JMRI #12269: DCC-EX: Add virtual LCD display

mstevetodd, 2023-08-11:
DCC-EX supports various LCD and OLED displays, and the strings sent to the display are echoed to JMRI in a series of "debug" messages, such as:
```
<* LCD2:Power Off *>
<* LCD3:Free RAM= 2374b *>
<* LCD4:192.168.0.173 *>
<* LCD5:PORT=2560 *>
<* LCD3:Free RAM= 2374b *>
```
I suggest that JMRI could add parsing for these messages and display them in a virtual LCD display window.

bobjacobsen, 2023-08-11:
I'd be happy to set up the window, connect it to the menus, etc in the jmrix.dccpp package. I don't know enough to do the parsing right 1st time and don't have access to the hardware to develop iteratively.  If somebody else will do the parsing code, I'll write the setup & display part.

bobjacobsen, 2023-08-11:
Initial support is available in the `dcc-ex-oled-display` branch in my BobJacobsen/JMRI repository. (See the [diffs here](https://github.com/JMRI/JMRI/compare/master...bobjacobsen:JMRI:dcc-ex-oled-display))

There a place marked with `// TODO` in each of `jmri.jmrix.dccpp.swing.display.DisplayFrame` and `jmri.jmrix.dccpp.swing.display.DisplayAction` that need some additional attention.

mstevetodd, 2023-08-11:
Here's a photo of the hardware version, from user mgar2112 on the DCC-EX Discord group:
[image]

mstevetodd, 2023-08-14:
I have asked the DCC-EX team to provide a specific message letter for these LCD messages, rather than parse them out of a debug message. They seemed open to the suggestion, but no progress on the change (yet). I'd like to wait a bit before giving up on that approach.

bobjacobsen, 2023-08-14:
Ok, thanks.  Happy to stand by.

Also, I think we can add a "5x8 LCD font" if you'd like the display to look like that image.

mstevetodd, 2023-09-30:
I'm still hoping one of the DCC-EX developers will work on that side.

mstevetodd, 2023-10-31:
Working on this.
Thanks, @bobjacobsen for the UI code! 
I'm still discussing the change with the DCC-EX code owners. My current suggestion is to use "@" as the opcode, and to broadcast the message over all streams instead of just USB. May need to add a "request" from JMRI to enable this. If so, I'll send when the lcd display is opened in JMRI.

mstevetodd, 2023-11-02:
@bobjacobsen 
Thanks again for the UI code!, it's included in my PR.
I would like some help (or suggestions) for improving the look. You mentioned a 5x8 LCD font? Can you advise further on that? Or even add it in after this PR completes?
I'd also like thoughts on changing the display dimensions. There are a number of LCD and OLED screens supported by DCC-EX, with varying rows and columns. Would it make sense to start with a small display, but dynamically resize it to fit the rows that are sent? Or something else?

bobjacobsen, 2023-11-02:
I'd be happy to add in the LCD font once this is merged.  It's a little bit tricky because it requires some infrastructure changes.

On appearance & sizing:  it's straightforward to size the window for the number of characters received so that it snugly fits.  It's harder to do the reverse, which is to resize the characters to fit the given size of the window. Which did you have in mind?

bobjacobsen, 2023-11-02:
The font I'd propose to use is [here](https://fonts2u.com/5x8-lcd-hd44780u-a02-regular.font)

bobjacobsen, 2023-11-03:
[image]

mstevetodd, 2023-11-03:
> On appearance & sizing: it's straightforward to size the window for the number of characters received so that it snugly fits. It's harder to do the reverse, which is to resize the characters to fit the given size of the window. Which did you have in mind?

Resizing the window, not the font size. Currently, DCC-EX supports displays from 2 rows x 16 columns up to 64 rows x 128 columns. Can the screen default to 2x16 but then auto-resize, adding more rows and columns if needed?

bobjacobsen, 2023-11-03:
> Currently, DCC-EX supports displays from 2 rows x 16 columns up to 64 rows x 128 columns. Can the screen default to 2x16 but then auto-resize, adding more rows and columns if needed?

Sort-of.  JMRI remembers the size and position of its windows. We can make the window size itself to its contents until the first time you move / resize / close it, after which that size and position will be remembered for the next time the window is drawn.

mstevetodd, 2023-11-03:
> We can make the window size itself to its contents until the first time you move / resize / close it, after which that size and position will be remembered for the next time the window is drawn.

That seems like a reasonable plan.

PR #12580 (2023-11-04): New DCC-EX Virtual LCD display
Closes #12269
closed 2023-11-04 by danielb987