JMRI/JMRI #12280: Problem with System Console "always on top"

jerryg2003, 2023-08-16:
Using 5.5.3, Windows 11, Java 11.0.16.1

Had System Console check box for "always on top."  Tried to do a delete of an ID Tag table entry and the Yes/No dialog box was hidden by the System Console.  Couldn't move any windows or even select a different window using Windows selector.  Couldn't deselect the check box on the System Console.  Finally had to abnormally shut down JMRI via the Task Manager.

I understand that "always on top" means always on top, but there has to be some way to get to a dialog box that is blocking any other action, doesn't there?

klk32003, 2023-08-17:
You can close the dialog box with alt+F4.

jerryg2003, 2023-08-18:
@klk32003 

Indeed, alt+F4 closed the dialog window.  I wasn't aware of this and in fact learned that "Alt + F4 is a Windows keyboard shortcut that completely closes the application you're using. It differs slightly from Ctrl + F4, which closes the current window of the application you're viewing."  Ctrl+F4 appears to have no effect.  I'll have to figure out a few places to put that into the documentation. 

Do Mac and Linux users have a similar way out?

klk32003, 2023-08-19:
This [site](https://helpdeskgeek.com/how-to/move-a-window-with-the-keyboard-in-windows/)  shows another, probably better  solution: ```Windows Key + Shift + right or left arrow```.
But the Windows Key might reduce the usage to Windows.

[Here](https://help.ubuntu.com/community/KeyboardShortcuts) are the KeyboardShortcuts for Ubuntu.
This site brings up another option: Alt + Space. This also works on my Windows 10 computer.

I cannot test a KeyboardShortcut on macOS. Maybe someone with a Mac could help.
Or with a software solution?

icklesteve, 2023-08-19:
Hi Folks,

I think the main issue in this case may be that the JDialog displaying the DeleteBeanWorker is set to Modal though without a JFrame.

Setting the JDialog to the JFrame holding the table, and Dialog.ModalityType.DOCUMENT_MODAL , and all starts working as I would expect.

While testing behaviours I've been manually fetching the JFrame from JmriJFrame , next step is to get the actual JFrame accessible to Dialogs which are triggered from BeanTableDataModels.

closed 2023-10-10 by jerryg2003