JMRI/JMRI #12282: Switchboard Text not fitting the button area

CliveRich, 2023-08-16:
Issue Type: __Bug__

Having set up the switches and routes using the built-in editor, I notice that the area for each button only shows all the text when in "button" mode. The "symbol" mode does not show all the text. The area of each button does not expand to fit the text, or the text is not reduced to fit the area of the button. This is the case no matter how large I make the switchboard window.  This is the same on any device using the web system to view the routes/switches.
How can I fix this.

PanelPro 5.4+R9005ecb4ac
OpenJDK 64-Bit Server VM OpenLogic 11.0.19+7-adhoc..jdk11u mixed mode
Windows 11 10.0 amd64

[image]
[image]
[image]
[image]
[image]

bobjacobsen, 2023-08-24:
I will take a look at this for the JMRI windows, but I don't know enough to work on the web page display.

mstevetodd, 2023-08-24:
@CliveRich or @bobjacobsen Can one of you attach an example panel file showing this issue?

CliveRich, 2023-08-25:
Thanks Bob.Much appreciated.RegardsCliveSent from my mobile device.

bobjacobsen, 2023-08-29:
PR #12304 has improved this for presentation on the native JMRI screen.  it would be useful to know if it also improved it when served to a web page.

CliveRich, 2023-10-14:
Thanks Bob for your help. The text size/text box relationship function does still not manage the text to image size correctly. I know nothing about HTML, and can't operate a self-help. I can use the button switch shape, as this does show the full text. It seem that no one else is bothered by this, as there would be a lot of "me too" responses. So  I think we will leave it there. 

This bug report can be closed.

closed 2023-10-14 by CliveRich

bobjacobsen, 2023-10-14:
Are you still seeing missing text on the JMRI computer's screen?  Or just on web pages?

If on the JMRI screen, could you provide your panel file (add a .txt to the name and attach it here) so I can work on this further?

CliveRich, 2023-10-14:
Bob,

It is only the webthrottle. All else is fine and workable. That is why I think it is an HTML function. As no one else has ever made a similar observation, I don’t think it is worth pursuing. Which is why I closed the submission.

Thanks.  

Regards

Clive