JMRI/JMRI #12283: Logic in the web page to update memory values

StCircumstance, 2023-08-18:
Cannot update memory values on a Control Panel when viewed as a web page.

StCircumstance, 2023-08-18:
It would be helpful to have all (4) types of memory items have the ability to be updated from the web page i.e. Input Box, Display, Spinner & Combo Box

mstevetodd, 2023-08-18:
Please attach a simple panel which includes examples of the types you are referring to.

StCircumstance, 2023-08-21:
Attached showing the 4 types. On the right is the panel running on the laptop running JMRI. the left is a tablet viewing the panel as a web page.
[image]

danielb987, 2023-10-31:
I have reopened this issue since it would be good if it could be fixed.

bobjacobsen, 2025-12-15:
It would also be great if one could edit the value in memory on a web-displayed Layout Editor panel.  

I've attached a file with a single LE panel with a single memory on it.

[MemoryOnLayoutEditorPanel.xml.txt](https://github.com/user-attachments/files/24165288/MemoryOnLayoutEditorPanel.xml.txt)

dsand47, 2026-01-20:
The main issue with adding the additional memory icon types to the LE toolbar is that the toolbar is very crowded.

[image]

One idea is to extend the context menu to include options to change the memory icon along with the supporting options for spinners and combo items.

This issue is being revisited due to a request from Dave Barraza.  However, he was referring to block content icons.  Since these are extended from memory icons, the same context menu approach could be applied to them.

mstevetodd, 2026-01-21:
> just allow the direct click-and-enter behavior on a web page?

Yes, that can be done, but I'm not sure about the appearance of that approach. See snip from OneOfEach.xml below. The "Memory" values give zero indication that it is an Input, where the "MemoryInput" clearly does. Browsers (by default) will make an input look like an input. So, I'd have to add code to "undo" the "proper" appearance,, just so the appearance of "display-only" values doesn't suddenly change for existing web panels.
I really prefer making the user choose input vs display-only when building the panel, and agree with @dsand47 recap of the UI issue.

[image]

mstevetodd, 2026-02-02:
Hopefully not, but I'll check it when you create/update the PR.

dsand47, 2026-02-02:
The PR has been updated with the BlockContentsInputIcon.