JMRI/JMRI #12287: Throttles are not disposed when Throttle window is closed

DeanCording, 2023-08-20:
I'm working on a new throttle controller interface and I noticed that when I close a Throttle window in PanelPro, the connected Throttle object doesn't get released and disposed of properly.

The log trace for creating a Throttle shows:

```
[java] 16:14:03,581 jmri.jmrix.AbstractThrottle           DEBUG - Adding property change jmri.jmrit.throttle.AddressPanel
     [java] 16:14:03,583 jmri.jmrix.AbstractThrottle           DEBUG - Speed Step Mode Change from:128 SS to:128 SS [AWT-EventQueue-0]
     [java] 16:14:03,583 jmri.jmrix.AbstractThrottle           DEBUG - Adding property change jmri.jmrit.throttle.ControlPanel
     [java] 16:14:03,584 jmri.jmrix.AbstractThrottle           DEBUG - Adding property change jmri.jmrit.throttle.FunctionPanel
     [java] 16:14:03,585 jmri.jmrix.AbstractThrottle           DEBUG - Adding property change jmri.jmrit.throttle.SpeedPanel
     [java] 16:14:03,586 jmri.jmrix.AbstractThrottle           DEBUG - Adding property change jmri.jmrit.throttle.ThrottlesTableModel@4aae45c3 [AWT-EventQueue-0]
     [java] 16:14:03,586 jmri.jmrix.AbstractThrottleManager    DEBUG - 1122(L) increased Use Size to 1 [AWT-EventQueue-0]
     [java] 16:14:03,586 jmri.jmrix.AbstractThrottleManager    DEBUG - Throttle 1122(L) now has 1 users [AWT-EventQueue-0]
     [java] 16:14:03,586 jmri.jmrix.AbstractThrottleManager    DEBUG - notifyThrottleKnown with zero-length propertyChangeListeners: 1122(L) [AWT-EventQueue-0]

```

So there are 5 property change listeners registered to the Throttle.  However, when the Throttle window is closed, the log trace is:

```
     [java] 16:20:47,634 jmri.jmrix.AbstractThrottle           DEBUG - Removing property change jmri.jmrit.throttle.ControlPanel
     [java] 16:20:47,634 jmri.jmrix.AbstractThrottle           DEBUG - remove listeners size is 4 [AWT-EventQueue-0]
     [java] 16:20:47,635 jmri.jmrix.AbstractThrottle           DEBUG - Removing property change jmri.jmrit.throttle.FunctionPanel
     [java] 16:20:47,635 jmri.jmrix.AbstractThrottle           DEBUG - remove listeners size is 3 [AWT-EventQueue-0]
     [java] 16:20:47,635 jmri.jmrix.AbstractThrottle           DEBUG - Removing property change jmri.jmrit.throttle.SpeedPanel
     [java] 16:20:47,635 jmri.jmrix.AbstractThrottle           DEBUG - remove listeners size is 2 [AWT-EventQueue-0]
     [java] 16:20:47,635 jmri.jmrix.AbstractThrottle           DEBUG - Removing property change jmri.jmrit.throttle.AddressPanel
     [java] 16:20:47,635 jmri.jmrix.AbstractThrottle           DEBUG - remove listeners size is 1 [AWT-EventQueue-0]
     [java] 16:20:47,636 jmri.jmrix.AbstractThrottleManager    DEBUG - AbstractThrottleManager.releaseThrottle: jmri.jmrix.mqtt.MqttThrottle@1697e368, jmri.jmrit.throttle.AddressPanel
     [java] 16:20:47,636 jmri.jmrix.mqtt.MqttThrottleManager   DEBUG - Dispose Throttle [AWT-EventQueue-0]
     [java] 16:20:47,636 jmri.jmrix.AbstractThrottleManager    DEBUG - AbstractThrottleManager.disposeThrottle: jmri.jmrix.mqtt.MqttThrottle@1697e368, jmri.jmrit.throttle.AddressPanel
     [java] 16:20:47,636 jmri.jmrix.AbstractThrottleManager    DEBUG - decrementUse called with listener jmri.jmrit.throttle.AddressPanel
     [java] 16:20:47,636 jmri.jmrix.AbstractThrottleManager    DEBUG - 1122(L) decreased Use Size to 0 [AWT-EventQueue-0]
     [java] 16:20:47,636 jmri.jmrix.AbstractThrottleManager    DEBUG - Throttle 1122(L) still has 1 active propertyChangeListeners registered to the throttle [AWT-EventQueue-0]
```

That is, only 4 of the listeners are removed - jmri.jmrit.throttle.ThrottlesTableModel is not removed.  This causes disposeThrottle in AbstractThrottleManager to fail and return false,  which in turn causes the implemented ThrottleManager to fail early before the Throttle is disposed of.

```
    public boolean disposeThrottle(DccThrottle t, ThrottleListener l) {
        log.debug("AbstractThrottleManager.disposeThrottle: {}, {}", t, l);

//        if (!active) log.error("Dispose called when not active");  <-- might need to control this in the sub class
        LocoAddress la = t.getLocoAddress();
        if (addressReleased(la, l)) {
            log.debug("Address {} still has active users", t.getLocoAddress());
            return false;
        }
        if (t.getPropertyChangeListeners().length > 0) {
            log.debug("Throttle {} still has {} active propertyChangeListeners registered to the throttle", t.getLocoAddress(), t.getPropertyChangeListeners().length);
            return false;
        }
        synchronized (this) {
            if (addressThrottles.containsKey(la)) {
                addressThrottles.remove(la);
                log.debug("Loco Address {} removed from the stack ", la);
            } else {
                log.debug("Loco Address {} not found in the stack ", la);
            }
        }
        return true;
    }
```

This problem means that for many controllers, throttles/addresses are not released by the controller if the user fails to release them before closing the Throttle window.

I've tried looking for the root cause of the issue, but it is too much of a rabbit hole to go down to try to understand the entire JMRI internals.

Lionel642, 2023-08-21:
>      [java] 16:14:03,586 jmri.jmrix.AbstractThrottle           DEBUG - Adding property change ***@***.*** [AWT-EventQueue-0]
>      [java] 16:20:47,636 jmri.jmrix.AbstractThrottleManager    DEBUG - AbstractThrottleManager.releaseThrottle: ***@***.***, jmri.jmrit.throttle.AddressPanel
>      [java] 16:20:47,636 jmri.jmrix.AbstractThrottleManager    DEBUG - AbstractThrottleManager.disposeThrottle: ***@***.***, jmri.jmrit.throttle.AddressPanel

Hello;

I'm on it, only happening when closing the window, I'll push a fix very
soon. Thank you for catching this.

Regards;
Lionel

DeanCording, 2023-08-28:
Testing with #12291.  The issue still happens when closing a throttle window but now also happens when you release a throttle using the Release button as well.

```
     [java] 17:26:45,178 jmri.jmrix.AbstractThrottle           DEBUG - removePropertyChangeListener(): Removing property change ControlPanel from 112(S) [AWT-EventQueue-0]
     [java] 17:26:45,179 jmri.jmrix.AbstractThrottle           DEBUG - removePropertyChangeListener(): listeners size is 4 [AWT-EventQueue-0]
     [java] 17:26:45,179 jmri.jmrix.AbstractThrottle           DEBUG - removePropertyChangeListener(): Removing property change FunctionPanel from 112(S) [AWT-EventQueue-0]
     [java] 17:26:45,179 jmri.jmrix.AbstractThrottle           DEBUG - removePropertyChangeListener(): listeners size is 3 [AWT-EventQueue-0]
     [java] 17:26:45,185 jmri.jmrix.AbstractThrottle           DEBUG - removePropertyChangeListener(): Removing property change SpeedPanel from 112(S) [AWT-EventQueue-0]
     [java] 17:26:45,186 jmri.jmrix.AbstractThrottle           DEBUG - removePropertyChangeListener(): listeners size is 2 [AWT-EventQueue-0]
     [java] 17:26:45,187 jmri.jmrix.AbstractThrottle           DEBUG - removePropertyChangeListener(): Removing property change AddressPanel from 112(S) [AWT-EventQueue-0]
     [java] 17:26:45,187 jmri.jmrix.AbstractThrottle           DEBUG - removePropertyChangeListener(): listeners size is 1 [AWT-EventQueue-0]
     [java] 17:26:45,187 jmri.jmrix.mqtt.MqttThrottleManager   DEBUG - releaseThrottle jmri.jmrix.mqtt.MqttThrottle@678cdc91 [AWT-EventQueue-0]
     [java] 17:26:45,362 jmri.jmrix.mqtt.MqttConsist           INFO  - Deactivating consist 112(S) [AWT-EventQueue-0]
     [java] 17:26:45,365 jmri.jmrix.mqtt.MqttThrottleManager   DEBUG - disposeThrottle jmri.jmrix.mqtt.MqttThrottle@678cdc91 [AWT-EventQueue-0]
     [java] 17:26:45,365 jmri.jmrix.AbstractThrottleManager    DEBUG - AbstractThrottleManager.disposeThrottle: jmri.jmrix.mqtt.MqttThrottle@678cdc91, jmri.jmrit.throttle.AddressPanel
     [java] 17:26:45,385 jmri.jmrix.AbstractThrottleManager    DEBUG - decrementUse called with listener jmri.jmrit.throttle.AddressPanel
     [java] 17:26:45,385 jmri.jmrix.AbstractThrottleManager    DEBUG - 112(S) decreased Use Size to 0 [AWT-EventQueue-0]
     [java] 17:26:45,385 jmri.jmrix.AbstractThrottleManager    DEBUG - Throttle 112(S) still has 1 active propertyChangeListeners registered to the throttle [AWT-EventQueue-0]
     [java] 17:26:45,385 jmri.jmrix.mqtt.MqttThrottleManager   ERROR - Dispose Throttle failed jmri.jmrix.mqtt.MqttThrottle@678cdc91 [AWT-EventQueue-0]
```

Lionel642, 2023-08-28:
I just tested, I can't reproduce, are you able to reproduce without creating a mqtt throttle, simply open a GUI throttle, acquire, release/close and check logs? (that's what I did here, looks ok). I must be missing something, let me know.

DeanCording, 2023-08-28:
The thought of implementing a train control system in Node Red did cross my mind but that would be akin to building a real train control system with hard wired relays - interesting project but I just want to play trains.

After seeing some of the 15-20 year old date stamps on code, I got a bit wary of delving too deep into the inner workings.  That said, the trains I'm working with are over 60 years old and I'm  writing a STM32 based controller for the non-DCC layout.  I looked at all of the other controller protocols and they all had ancient propriety hardware peculiarities baked into them, so I would end up having to write an emulator of some 15+ year old system.  MQTT provides a nice hardware agnostic interface that opens up  many possibilities such as IoT integration, external dashboards, WiFi sensors and activators, ....

Lionel642, 2023-08-31:
again, thanks, I'm testing with Loconet where it's working, but switching to DCC+ I was able to reproduce, and that's because I used the == comparator on LocoAddress objects instead of .equals(). Pushing now.

DeanCording, 2023-09-02:
Yay, that's nailed it.  Throttles are now correctly disposed.  Thanks for your efforts on this.

DeanCording, 2023-09-09:
This issue has reappeared.  Possibly something related to @icklesteve changes in #12390

DeanCording, 2023-09-09:
Log file:

Throttle creation:

```
[java] 22:48:59,337 jmri.jmrix.AbstractThrottleManager    DEBUG - LocoAddress 1 has not been created before [AWT-EventQueue-0]
     [java] 22:48:59,337 jmri.jmrix.AbstractThrottleManager    DEBUG - After request in ATM: 0 [AWT-EventQueue-0]
     [java] 22:48:59,338 jmri.jmrix.AbstractThrottleManager    DEBUG - list of WaitingThrottles is empty: 1(S); [jmri.jmrix.AbstractThrottleManager$WaitingThrottle@4c17c158] [AWT-EventQueue-0]
     [java] 22:48:59,338 jmri.jmrix.AbstractThrottleManager    DEBUG - calling requestThrottleSetup() [AWT-EventQueue-0]
     [java] 22:48:59,338 jmri.jmrix.mqtt.MqttThrottleManager   DEBUG - Requesting Throttle: 1(S) [AWT-EventQueue-0]
     [java] 22:48:59,342 jmri.jmrix.mqtt.MqttAdapter           DEBUG - Subscribed : "state/cab/1/throttle" [AWT-EventQueue-0]
     [java] 22:48:59,344 jmri.jmrix.mqtt.MqttAdapter           DEBUG - Subscribed : "state/cab/1/direction" [AWT-EventQueue-0]
     [java] 22:48:59,345 jmri.jmrix.mqtt.MqttAdapter           DEBUG - Subscribed : "state/cab/1/function/#" [AWT-EventQueue-0]
     [java] 22:48:59,353 jmri.jmrix.mqtt.MqttThrottle          DEBUG - sent address 1 speed 0 [AWT-EventQueue-0]
     [java] 22:48:59,353 jmri.jmrix.mqtt.MqttAdapter           DEBUG - Message delivered [MQTT Call: ri525400F4CDF13f453f7fM]
     [java] 22:48:59,359 jmri.jmrix.mqtt.MqttThrottle          DEBUG - sent address 1 direction FORWARD [AWT-EventQueue-0]
     [java] 22:48:59,360 jmri.jmrix.mqtt.MqttAdapter           DEBUG - Message delivered [MQTT Call: ri525400F4CDF13f453f7fM]
     [java] 22:48:59,360 jmri.jmrix.mqtt.MqttThrottle          DEBUG - MqttThrottle constructor called for address 1(S) [AWT-EventQueue-0]
     [java] 22:48:59,360 jmri.jmrix.AbstractThrottleManager    DEBUG - notifyThrottleKnown for 1(S) [AWT-EventQueue-0]
     [java] 22:48:59,360 jmri.jmrix.AbstractThrottleManager    DEBUG - Address 1(S) doesn't already exists so will add [AWT-EventQueue-0]
     [java] 22:48:59,361 jmri.jmrix.AbstractThrottleManager    DEBUG - Notify listener 1 of 1 [AWT-EventQueue-0]
     [java] 22:48:59,361 jmri.jmrix.AbstractThrottle           DEBUG - addPropertyChangeListener(): Adding property change AddressPanel to 1(S) [AWT-EventQueue-0]
     [java] 22:48:59,362 jmri.jmrix.AbstractThrottle           DEBUG - addPropertyChangeListener(): throttle: 1(S) listeners size is 1 [AWT-EventQueue-0]
     [java] 22:48:59,364 jmri.jmrix.AbstractThrottle           DEBUG - Speed Step Mode Change from:128 SS to:128 SS [AWT-EventQueue-0]
     [java] 22:48:59,365 jmri.jmrix.AbstractThrottle           DEBUG - addPropertyChangeListener(): Adding property change ControlPanel to 1(S) [AWT-EventQueue-0]
     [java] 22:48:59,365 jmri.jmrix.AbstractThrottle           DEBUG - addPropertyChangeListener(): throttle: 1(S) listeners size is 2 [AWT-EventQueue-0]
     [java] 22:48:59,366 jmri.jmrix.AbstractThrottle           DEBUG - addPropertyChangeListener(): Adding property change FunctionPanel to 1(S) [AWT-EventQueue-0]
     [java] 22:48:59,366 jmri.jmrix.AbstractThrottle           DEBUG - addPropertyChangeListener(): throttle: 1(S) listeners size is 3 [AWT-EventQueue-0]
     [java] 22:48:59,370 jmri.jmrix.AbstractThrottle           DEBUG - addPropertyChangeListener(): Adding property change SpeedPanel to 1(S) [AWT-EventQueue-0]
     [java] 22:48:59,370 jmri.jmrix.AbstractThrottle           DEBUG - addPropertyChangeListener(): throttle: 1(S) listeners size is 4 [AWT-EventQueue-0]
     [java] 22:48:59,371 jmri.jmrix.AbstractThrottle           DEBUG - addPropertyChangeListener(): Adding property change ThrottlesTableModel to 1(S) [AWT-EventQueue-0]
     [java] 22:48:59,371 jmri.jmrix.AbstractThrottle           DEBUG - addPropertyChangeListener(): throttle: 1(S) listeners size is 5 [AWT-EventQueue-0]
     [java] 22:48:59,371 jmri.jmrix.AbstractThrottleManager    DEBUG - 1(S) increased Use Size to 1 [AWT-EventQueue-0]
     [java] 22:48:59,372 jmri.jmrix.AbstractThrottleManager    DEBUG - Throttle 1(S) now has 1 users [AWT-EventQueue-0]
     [java] 22:48:59,372 jmri.jmrix.AbstractThrottleManager    DEBUG - notifyThrottleKnown with zero-length propertyChangeListeners: 1(S) [AWT-EventQueue-0]
```

Throttle disposal
```
    [java] 22:49:05,085 jmri.jmrix.AbstractThrottle           DEBUG - removePropertyChangeListener(): Removing property change ControlPanel from 1(S) [AWT-EventQueue-0]
     [java] 22:49:05,085 jmri.jmrix.AbstractThrottle           DEBUG - removePropertyChangeListener(): throttle: 1(S) listeners size is 4 [AWT-EventQueue-0]
     [java] 22:49:05,085 jmri.jmrix.AbstractThrottle           DEBUG - removePropertyChangeListener(): Removing property change FunctionPanel from 1(S) [AWT-EventQueue-0]
     [java] 22:49:05,086 jmri.jmrix.AbstractThrottle           DEBUG - removePropertyChangeListener(): throttle: 1(S) listeners size is 3 [AWT-EventQueue-0]
     [java] 22:49:05,092 jmri.jmrix.AbstractThrottle           DEBUG - removePropertyChangeListener(): Removing property change SpeedPanel from 1(S) [AWT-EventQueue-0]
     [java] 22:49:05,092 jmri.jmrix.AbstractThrottle           DEBUG - removePropertyChangeListener(): throttle: 1(S) listeners size is 2 [AWT-EventQueue-0]
     [java] 22:49:05,093 jmri.jmrix.AbstractThrottle           DEBUG - removePropertyChangeListener(): Removing property change ThrottlesTableModel from 1(S) [AWT-EventQueue-0]
[... cut ...]
     [java] 22:49:05,095 jmri.jmrix.AbstractThrottleManager    DEBUG - addressReleased still has at least one listener [AWT-EventQueue-0]
     [java] 22:49:05,095 jmri.jmrix.AbstractThrottleManager    DEBUG - Address 1(S) still has active users [AWT-EventQueue-0]
     [java] 22:49:05,096 jmri.jmrix.mqtt.MqttThrottleManager   ERROR - Dispose Throttle failed jmri.jmrix.mqtt.MqttThrottle@21b3b9a7 [AWT-EventQueue-0]
     [java] 22:49:05,096 jmri.jmrix.mqtt.MqttThrottleManager   DEBUG - releaseThrottle jmri.jmrix.mqtt.MqttThrottle@21b3b9a7 [AWT-EventQueue-0]
```

closed 2023-09-09 by DeanCording

danielb987, 2023-09-10:
Was it intended to close this issue?

DeanCording, 2023-09-10:
Yes,  the throttle release logic generates a spurious error message which shows up in log at the Error level, but when you log at the Debug level you can see that it's just how it works.