JMRI/JMRI #12296: no product id code for Dietz decoders

FancyRedCat, 2023-08-23:
I bought recently a decoder from Dietz. That's why I try to write a new decoder definition.
There are no Dietz decoder definition in JMRI yet, so JMRI cannot identify the product id code from Dietz decoders.

The manufactor code in CV8 is 115
and the product id is stored in CV128.

There are no english manuals available, only german: https://www.d-i-e-t-z.de/aktuelle-anleitungen.html
But I wrote an email to Dietz and they confirmed the CV128.

bobjacobsen, 2023-08-23:
Thank you.  PR #12297 has the fix for this.

closed 2023-08-24 by danielb987

bobjacobsen, 2023-08-24:
PR #12297 has been merged, so this fix will be available in a development build early tomorrow.  You can pick up installers at https://builds.jmri.org/jenkins/job/development/job/packages/