JMRI/JMRI #12305: QUERY Button on Sensor Table causes Unknown

slolitzky, 2023-08-25:
Issue Type: __Bug__

I have implemented block sensors using C/MRI and an Arduino UNO.  The sensor status is returned back correctly when JMRI first comes up and if there is later a state change in one or more sensors.  If I manually change a Sensor state (in the sensor table) so that I can check that Block Occupancy is working ok,  the Sensor state will change to the new state that I have toggled.  It will stay that way even if the Arduino is still returning the previous state from the hardware (as expected).

HOWEVER when I use the QUERY button on the Sensor table, the sensor state is not updated from hardware status, instead the State says "Unknown" instead of either Active or Inactive.  If I use the FORGET button it also will set a sensor state to "Unknown".

This problem also occurs with the Turnout Sensors.

PanelPro 5.4+R9005ecb4ac
Java HotSpot(TM) 64-Bit Server VM Oracle Corporation 11.0.18+9-LTS-195 mixed mode
Windows 11 10.0 amd64

[systemInfo.txt](https://github.com/JMRI/JMRI/files/12434352/systemInfo.txt)
[profile.zip](https://github.com/JMRI/JMRI/files/12434353/profile.zip)
[session.log](https://github.com/JMRI/JMRI/files/12434354/session.log)

bobjacobsen, 2023-08-25:
This seems to be a bug with the Query button (The Forget button is doing what it's supposed to do).  I have a tentative fix for this, but don't have any C/MRI hardware to test with.  Would you please download an installer from

https://builds.jmri.org/jenkins/job/testreleases/job/cmri-update-button/

and see if that fixes the Query button problem?  Thanks.

slolitzky, 2023-08-25:
Thanks.  I  tested it with both the block occupancy sensors and my turnout sensors.  It appears to work now.  
Seth

closed 2023-08-25 by slolitzky