JMRI/JMRI #12314: Can't launch JMRI programs

RogerFearnside, 2023-08-27:
System: Dell Venue 11 Pro. Intel M-Sy10c CPU. 4Gb Ram. Windows 10 Pro 22H2 AZUL Java 11 JRE
I've had JMRI v5.2 running on this system, but became unstable. Decided to remove it and install v5.4.
PanelPro, DecoderPro, SoundPro won't start.
Error is: No Command line parameters with LaunchJMRI.exe 
I start the programs from the created desktop icons.. Checked the target properties and all is as it should be.
JMRI install test says "There is something wrong with invoking JMRI" InstallTest log attached:
[error.txt](https://github.com/JMRI/JMRI/files/12448038/error.txt)

I've tried versions 5.0 and 5.2 but still the same

klk32003, 2023-09-10:
It's very difficult to analyze an "unstable" computer system. Maybe that's the reason why there are no responses so far. Probably it is rather the wrong place for this question. The jmriusers forum is the better choice.

My guess is that it is an installation issue due to a Java problem.

https://www.jmri.org/install/WindowsNew.shtml

RogerFearnside, 2023-09-12:
Thank you
I've installed JMRI on many systems with no problems. Maybe I'll try a
reinstall of windows. The PC is dedicated to JMRE/Arduino so not much to
lose but a few config files.
Your input appreciated.
Roger

RogerFearnside, 2023-10-31:
Hi
Sorry.I've not been back to you. I've just found the answer. (Hope it will
help others) It was a corrupted JMRIconfig file. What threw me was that
when I refreshed my Windows installation, removing all apps, it did not
delete my data, including the config file!
Re-installed JMRI with a new config, and all works (apart from losing my
settings).
All is well now.
Thank you
Best wishes
Roger Fearnside

closed 2023-11-02 by bobjacobsen