JMRI/JMRI #12317: Add DCC-EX Support for Fkeys up to 68 to JMRI Throttle

mstevetodd, 2023-08-27:
A Discord user has a decoder which shows 31 Function Keys in the Labels & Media section, but those keys do not show in the JMRI Throttle window.
I see recent PRs which seem to have added this support for Lenz and Digitrax connections, we need to add this support for DCC-EX as well.

bobjacobsen, 2023-08-27:
This would involve implementing the `sendFunctionGroup6` et al methods of AbstractThrottle in the DCC-EX throttle class.

mstevetodd, 2023-08-27:
@bobjacobsen Thanks, but I had already rewritten the send to no longer use the groups for DCC-EX. So the change was simply to include totalFunctions in the AbstractThrottle constructor. PR on the way.