JMRI/JMRI #12337: Locos controlled by TCS UWT will not stop on JMRI wiThrottle server

newHeiko, 2023-09-01:
From the JMRI users list: https://groups.io/g/jmriusers/message/220524

Quoting Ross:

> The problem being discussed is “When brining the throttle down to the lower numbers and then to zero the loco does not stop.”  The decoders are responding but apparently are not receiving a ‘0’ speed command in some situations.
> 
> I’ve experienced this phenomena with UWT-100, UWT-50e and UWT-50p throttles on 3 different Digitrax DCC systems.  I’ve also experienced it on a large NCE DCC system.  All of these were using JMRI WiThrottle server.  

Quoting myself:

> As I recall it: After receiving a "Speed step 1" command from the UWT, JMRI sends "something" to loconet and receives back what it incorrectly interprets as a "Speed step 0" response.
>
> This wrong "Speed step 0" response gets forwarded to the TCS UWT. Which then (understandably) does not send a "Speed step 0" command at all, because it believes the loco is already at standstill.
>
> This is true for the 28 speed steps <-> Loconet translation, see https://jmri-developers.groups.io/g/jmri/message/7649 and I wouldn't be surprised to see it also on the 14 speed steps, though I haven't checked.

I've added this issue so it doesn't get lost again and may see if I can work on something.

Heiko

newHeiko, 2023-09-01:
Question: Should we aim to achieve int x == intSpeed(floatSpeed(x)) for all x in 0..126 or do we only care about 0 == intSpeed(floatSpeed(0)) ?

Heiko

newHeiko, 2023-09-01:
> Question: Should we aim to achieve int x == intSpeed(floatSpeed(x)) for all x in 0..126 or do we only care about 0 == intSpeed(floatSpeed(0)) ?

Nevermind. Thinking about this, we can't do that, since for 14/28 speed steps we don't want to send all the integers in 0..126 to loconet but just a select few. I might try to achieve that for those specific speed steps though.

Heiko

danielb987, 2023-09-01:
@newHeiko @bobjacobsen 
Do we really need to do this? For me, it seems that it would be better to add a method `void setIntSpeedSetting(int speed)` to `jmri.Throttle` with a default implementation in `jmri.jmrix.AbstractThrottle` and an overridden method in `LocoNetThrottle`.

If a new method `getNumIntSpeedSteps()` is added, that method could tell the caller how many speed steps `setIntSpeedSetting(int speed)` handles. The default would be 100, which means `float speed = intSpeed/100.0`. But for LocoNetThrottle, this would be 128 I think.

The code in `jmri.jmrix.AbstractThrottle` would look something like this:
```
void setIntSpeedSetting(int speed) {
    setSpeedSetting(speed/100.0);
}

int getIntSpeedSetting() {
    ...
}

int getNumIntSpeedSteps() {
    return 100;
}
```

bobjacobsen, 2023-09-02:
I think the problem is the round-trip from int to float and back to int.  There are apparently values where that doesn't get you back to the same value, particularly around small values of speed and for different speed step settings.

IIRC, the original method was to a floating calculation of (intSpeed+0.5)/numSteps to get to float.  That's a reliable conversion back to numSteps via a floor(..) operation because of the 0.5 added in. 

Unfortunately, that gives bad results when e.g. calculating something in 14 or 28 steps and then interpreting that in terms of 100 (Digitrax) or 128 (really 127) DCC steps because the 0.5 offset of a 14-step is quite large in 128 step space.  Most importantly, it doesn't interpret intSpeed 0 as float speed 0 when converting step size.

The code has been modified since then, and I'm not sure how it works numerically now.  

I don't know whether building in native integer support would help that or not.

newHeiko, 2023-09-02:
I don't think that's not going to help the issue (plus it's probably going to reintroduce rounding errors where setIntSpeedSetting(getIntSpeedSetting()) will change the speed).

My test case will be, as mentioned on jmri-developers a year ago:

`
wiThrottle client "V1" -> JMRI wiThrottle server "1/126f" -> JMRI LocoNetThrottle (function intSpeed) "12" -> Loconet speed message "12" -> Loconet slot "12" -> JMRI LocoNetThrottle (function floatSpeed) "0f" -> JMRI wiThrottle server "V0" -> wiThrottle client.
`

I'm pretty sure that's what's causing the TCS throttles' misbehavior. Since I have the feeling (from reading the comments in LocoNetThrottle.floatSpeed) that 12 should in fact be 0, that's where I'll do the change (and also check AbstractThrottle and the other inheriting objects).

Heiko

newHeiko, 2023-09-02:
> Since I have the feeling (from reading the comments in LocoNetThrottle.floatSpeed) that 12 should in fact be 0, that's where I'll do the change (and also check AbstractThrottle and the other inheriting objects).

Though if that comment in LocoNetThrottle.floatSpeed "//Value less than 15 is in the stop/estop range bracket" is true, the loco shouldn't move on loconet speed 12. That's probably different loconet speed -> DCC speed mappings of different command stations once more.

Heiko

bobjacobsen, 2023-09-06:
Dog Johnson has provided some test info and a LocoNet log in a [jmriusers post](https://groups.io/g/jmriusers/message/220779).

newHeiko, 2023-09-06:
Yes. That confirms what I thought and also that our assumption in https://github.com/JMRI/JMRI/blob/cde2ab07166c87a8cb1fca5c07b08d42a90548f3/java/src/jmri/jmrix/loconet/LocoNetThrottle.java#L131C75-L131C75 is plain wrong for the command station Doug is using. This was introduced about 10 years ago in https://github.com/JMRI/JMRI/commit/f636e20d240ac42c9e9d0d38410d466ce0f92b5d and https://github.com/JMRI/JMRI/commit/b45b88f9f55f169b6a8905c6fc798a8991e05372 and has already been proven to be wrong for Intellibox 1, see https://github.com/JMRI/JMRI/commit/8ffc812b5d4c894a4c3e9601da366eedf9b82ddb

I won't fix that (yet) but only make sure we never send anything below 15 to loconet and check how this bug can happen on NCE systems.

For a full WiThrottle speed <-> loconet speed <-> DCC speed rewrite, I'll probably build a little DCC protocol analyzer and ask someone in my club with a lot of DCC command stations to do a complete test. That's a longer project ;) and will leave us with a lot of command-station specific behavior.

Heiko

newHeiko, 2023-09-17:
So, I think I have this mostly covered. There's a lot of places where non-zero input speeds can turn into zero output and as commented in AbstractThrottle, that's not desireable. But most of them have no direct relevance to this issue.

One central change I made is in MultiThrottleController.sendCurrentSpeed - everything else is split all over all the Throttle implementations.

I adapted as I think is necessary to fix this issue:

Pr2Throttle
LocoNetThrottle

I checked and think they should be correct wrt to this issue:
Ib1Throttle
Ib2Throttle (inherits from LocoNetThrottle)
CbusThrottle (uses AbstractThrottle.intSpeed)

I'll try to get those commits into one PR that should fix this issue and add a second PR for all the following...

Heiko

I added non-zero -> non-zero checks and replaced (int) (maxSpeedStep * fSpeed) with Math.round(maxSpeedStep * fSpeed) to align with AbstractThrottle, but they should not have been affected by this issue:

EcosDccThrottle
MrcThrottle (I'm guessing this does not handle emergency stop readback from hardware correctly, though @bobjacobsen @kevin-dickerson see my comment in line 273)
TamsThrottle (though I'm sure something is wrong in TamsThrottle.floatSpeed in the 28 speed step code path @janboen see my comment in line 256)
MqttThrottle
MarkinThrottle
SprogCSThrottle
SprogThrottle (side note: SprogThrottle and SprogCSThrottle could maybe inherit from one another instead of existing side by side @crosland )
Mx1Throttle
DCCppThrottle (in DCCppMessage.makeSpeedAndDirectionMsg)
XpaThrottle (though I'm not sure the 127 maximum is correct since it's not handling estop but incrementally running 0...127 @pabender )

I don't see how this bug affects:
NceThrottle (there is no Hardware speed -> JMRI speed codepath that I can see)
SCRPThrottle (same)
AbstractThrottle (same)
DebugThrottle (no float->int conversion in setSpeedSetting, no Hardware speed -> JMRI speed codepath that I can see)

I honestly don't understand:
Z21XNetThrottle
ElilteXNetThrottle
XNetThrottle
RocoXNetThrottle (inherits from XNetThrottle wrt to this issue)
OlcbThrottle

I ignored:
SerialThrottle

pabender, 2023-09-18:
> XpaThrottle (though I'm not sure the 127 maximum is correct since it's not handling estop but incrementally running 0...127 @pabender )

That should be right for the XPA.  The XPA code isn’t dealing with NMRA DCC packets directly.  The XPA code is sending a series of touch tone phone tones ( using a modem ) to the XPA, and the XPA then deals with The appropriate throttle instructions on the XpressNet.

> I honestly don't understand:
> 
> Z21XNetThrottle
> ElilteXNetThrottle
> XNetThrottle
> RocoXNetThrottle (inherits from XNetThrottle wrt to this issue)
Theoretically, all of those should inherit from XNetThrottle.

But what don’t you understand about them?

Is it the extra 27 speed step mode?

They probably do the calculations differently than the way Bob described.

pabender, 2023-09-18:
Sent from my iPhone

> On Sep 17, 2023, at 9:36 PM, Paul Bender ***@***.***> wrote:
> 
> 
>> XpaThrottle (though I'm not sure the 127 maximum is correct since it's not handling estop but incrementally running 0...127 @pabender )
> 
> That should be right for the XPA.  The XPA code isn’t dealing with NMRA DCC packets directly.  The XPA code is sending a series of touch tone phone tones ( using a modem ) to the XPA, and the XPA then deals with The appropriate throttle instructions on the XpressNet.

I’ll also add, the XPA doesn’t get any feedback about what the current step is from the hardware.

pabender, 2023-09-18:
@newHeiko I had a look at the two PRs you have out there, and there is a fundamental problem with them, namely that checking to ensure that a floating point number is equal to zero in any computer program is a bad idea ( this is due completely to the representation of the numbers, and not the mathematical operations involved )

i.e. you should never say ( for a floating point number X )
X==0.0
Because it likely never will be exactly equal to 0.0. ( and it’s worse if you use integer 0… because there has to be a conversion involved in the calculation )

What you should use is:

abs(X-0.0) < tol

where tol is your tolerance.

Your fix says that whenever the floating point representation of the speed is greater than zero, then a nonzero speed should be sent, but fundamentally that is saying the only way for the speed to actually be 0 is for the floating point representation of the number to actually be 0, and that’s not something we can count on.

So, instead of saying 
this.speedSetting > 0

you really need to say:
abs(this.speedSetting -0.0 ) > tol

This is also the reason why that PR2 test failed on you.  That small positive number was dreamed previously to be close enough to zero that it should be treated as zero, but your PR is forcing it to not be treated as a nonzero number.

newHeiko, 2023-09-18:
> XpaThrottle (though I'm not sure the 127 maximum is correct since it's not handling estop but incrementally running 0...127 @pabender )
> That should be right for the XPA. The XPA code isn’t dealing [...]

Thanks, then the 127 is correct. Also thank you for confirming there is no readback, then this issue probably won't hit them anyway.

> I honestly don't understand: Z21XNetThrottle ElilteXNetThrottle XNetThrottle RocoXNetThrottle (inherits from XNetThrottle wrt to this issue)
> Theoretically, all of those should inherit from [...]

I got lost somewhere following the speed value down from one function to the next to mixing it with extra bits etc.

Heiko

newHeiko, 2023-09-18:
> @newHeiko I had a look at the two PRs you have out there, and there is a fundamental problem with them, namely that checking to ensure that a floating point number is equal to zero in any computer [...]

While that's true for general floating point comparisons, I seem to recall comparing to 0.0 is OK in these cases. I looked that up when I changed AbstractThrottle.intSpeed two years ago, so I'm a little murky about the details, but it went something like: We only ever set float speed by multiplying an integer speed with a float 1/speedSteps value, and if that integer is zero, the float comparison to 0.0 is guaranteed to work. I can go back and compare to a threshold again, but then we'll be back to the "any non-zero float speed should move the loco" discussion. It'll still be better then the old implicit threshold.
 
> This is also the reason why that PR2 test failed on you. That small positive number was dreamed previously to be close enough to zero that it should be treated as zero, but your PR is forcing it to not be treated as a nonzero number.

No, that test fails because that small number previously fell between the cracks of "is zero" (-> should return 0) and "is first speed step" (-> should return 16) and the test was written after the code :) In fact, that's the base of this issue here.

Heiko

crosland, 2023-09-18:
> SprogCSThrottle SprogThrottle (side note: SprogThrottle and SprogCSThrottle could maybe inherit from one another instead of existing side by side @crosland )

That's probably true, but I am maxed out at the moment.

My question is, what has changed? I have never seen, nor heard of, an issue with achieving zero speed with a SPROG.

What is going to change? It makes me nervous, given all the discussion over this a while back.

pabender, 2023-09-19:
> > This is also the reason why that PR2 test failed on you. That small positive number was dreamed previously to be close enough to zero that it should be treated as zero, but your PR is forcing it to not be treated as a nonzero number.

> While that's true for general floating point comparisons, I seem to recall comparing to 0.0 is OK in these cases. I looked that up when I changed AbstractThrottle.intSpeed two years ago, so I'm a [...]

This might be true if we are always using the same speed step range, but in this case we aren’t because the WiThrottle server always assumes 128 speed step mode.

What needs to be guaranteed is that when you convert 

WiThrottle protocol speed -> float speed in -> hardware speed -> float speed out -> WiThrottle protocol speed

you always get the same speed back… regardless of what speed step mode the hardware is using.

> No, that test fails because that small number previously fell between the cracks of "is zero" (-> should return 0) and "is first speed step" (-> should return 16) and the test was written after the [...]

The mathematician in me ( who taught numberical methods for more than a few years ) says that isn’t right.  There are always “cracks” as you say.

The mathematical explanation is because we are dealing with discrete representations of continuous values, each value is actually representative of a range of values that should all be represented by the same speed step in this case. 

if we consider the 128 speed step case, the range is:
$$\left(\frac{n}{126}-\frac{1}{252} , \frac{n}{126}+\frac{1}{252}\right]$$

where n is the speed step.

since we use negative numbers for emergency stop, zero is actually represented by the range

$$\left(0,\frac{1}{252}\right]$$

Similar ranges exist for the other speed step modes, and necessarily, the ranges for 14 and 28 speed steps are larger than the ranges for 128 speed steps.

But the question, going back to an earlier statement, is are there any cases where the conversion:

WiThrottle protocol speed -> float speed in -> hardware speed -> float speed out -> WiThrottle protocol speed

fails to start and end with the same value.

of course, this isn’t new.  Bob already stated this above.

https://github.com/JMRI/JMRI/issues/12337#issuecomment-1703538373

newHeiko, 2023-09-19:
First of all, Paul, let me thank you again for engaging so deep in this discussion. I appreciate it, even if I partly disagree.

> This might be true if we are always using the same speed step range, but in this case we aren’t because the WiThrottle server always assumes 128 speed step mode.

I don't understand how that statement is related to float == comparison issues. The quantization is another topic (IMHO) that I'll get back to further down.

> What needs to be guaranteed is that when you convert
> 
> WiThrottle protocol speed in -> float speed in -> hardware speed -> float speed out -> WiThrottle protocol speed out
> 
> you always get the same speed back… regardless of what speed step mode the hardware is using.

I agree with you for 128 speed steps (in fact, that was what I fixed in #9464 a couple of years back for loconet). However, I think that's not generally possible for other speed step settings, because we map i.e. [0..126] -> [0..28] but can't invert (mathematically speaking) that mapping.

We do need to make sure that WiThrottle 0 in -> float 0 -> hardware speed 0 -> float speed 0 -> WiThrottle 0 out and (I think) WiThrottle >0 in -> float >0 -> hardware speed >0 -> float speed >0 -> WiThrottle >0 out.

The cause of this issue is: float 1/126 -> loconet speed 12 -> float speed 0 for 28 speed steps. I fixed the first conversion.

> > > This is also the reason why that PR2 test failed on you. That small positive number was dreamed previously to be close enough to zero that it should be treated as zero, but your PR is forcing it to not be treated as a nonzero number.
> > 
> > 
> > No, that test fails because that small number previously fell between the cracks of "is zero" (-> should return 0) and "is first speed step" (-> should return 16) and the test was written after the code :) In fact, that's the base of this issue here.
> 
> The mathematician in me ( who taught numerical methods for more than a few years ) says that isn’t right. There are always “cracks” as you say.
> 
> The mathematical explanation is because we are dealing with discrete representations of continuous values, each value is actually representative of a range of values that should all be represented by the same speed step in this case.
> 
> if we consider the 128 speed step case, the range is: $(\frac{n}{126} - \frac{1}{252}, \frac{n}{126} + \frac{1}{252}]$
> 
> where n is the speed step.

It could also be $[\frac{n}{126} - \frac{1}{252}, \frac{n}{126} + \frac{1}{252})$ but that doesn't make a functional difference in our use cases. So for most speed steps, yes, I agree.

> since we use negative numbers for emergency stop, zero is actually represented by the range
> 
> $(0,\frac{1}{252}]$

It should include 0, shouldn't it? (Or are there different mathematical notations for inclusive (square) and exclusive (round) brackets than I'm used to?)

Anyhow, that's what I'd have thought as well, but the discussion around #9464 / #9462 came to a different consensus: https://jmri-developers.groups.io/g/jmri/message/5242: The first speed step should be $(0, \frac{n}{126} + \frac{1}{252})$ (again, the right end could also be inclusive) leaving only 0 for speed step 0. I was working under the impression of that old consensus here.

I also now, on re-checking the old Pr2Throttle code, see that I mixed it up with LocoNetThrottle, so the crack I was talking about (loconet speed 0 for fSpeed == 0, loconet speed 16 (what we intend to send as first speed step) for fSpeed >= 1/28 and some unintended intermediate loconet steps in between) isn't present in Pr2Throttle. Sorry for that confusion.

Still, it's probably affected by this issue because it also outputs loconet speeds of 12...15 for fSpeed in $[0, \frac{1}{28})$ and reads them back as 0. This is based on some assumption that Loconet command stations send DCC 0 for loconet <15, but as this issue shows that's not generally true.

Heiko

newHeiko, 2023-09-19:
> > SprogCSThrottle SprogThrottle (side note: SprogThrottle and SprogCSThrottle could maybe inherit from one another instead of existing side by side @crosland )
> 
> That's probably true, but I am maxed out at the moment.

Fair. I also have bigger fish to fry.

> My question is, what has changed? I have never seen, nor heard of, an issue with achieving zero speed with a SPROG.
> 
> What is going to change? It makes me nervous, given all the discussion over this a while back.

I'm trying to have all the throttles behave the same, matching AbstractThrottle.intSpeed:

- for internal speed == 0, output 0
- for internal speed > 0 but < 1/126 + 1/252, output speed step 1
- for internal speed >= n/126 - 1/252 but < n/126 + 1/252, output speed step n
- for internal speed >= 126/126 - 1/252, output 126

(for all the different speed step settings - loconet in particular is weird because "speed step 1" translates to "loconet speed 16" for some command stations)

Before, most (including SprogCSThrottle) did
- for internal speed < 1/126, output 0
- for internal speed >= n/126 but < (n+1)/126, output speed step n
- for internal speed >= 126/126 output 126

That involved replacing (int) (fSpeed * 126) with Math.round(fSpeed * 126) (rounding up and down instead of just down) and adding the if(fSpeed > 0 && speed = 0) { speed = 1 } check.

That was just for consistency, because in the cause of fixing this issue, I stumbled over the fact that all those throttles were doing tiny things differently from one another and from AbstractThrottle. It's not necessary to fix this issue. Which is why I split it into a second PR.

Heiko

pabender, 2023-09-19:
> > since we use negative numbers for emergency stop, zero is actually represented by the range
> > (0,1/252]
> 
> It should include 0, shouldn't it? (Or are there different mathematical notations for inclusive (square) and exclusive (round) brackets than I'm used to?)

Oh, you're right, I forgot to change it to match the range, they should both be square brackets.  My error on the notation.

I'm still thinking about the math of all of this.  I'm concerned that by forcing all small non-zero speeds sent to the throttle to send speed step 1 in the currently selected speed range that you're going to change the behavior of the JMRI on screen and web throttles.  In other words, is moving the on screen throttle 1% sufficient movement for a locomotive running in 28 or 14 speed step mode to need to start moving?  I don't personally think it should, but your changes would force that.

Ideally, you would make whatever fix you need to only in the WiThrottle code, which may mean you need to make it aware of the current speed step setting of the underlying throttle.

newHeiko, 2023-09-19:
> In other words, is moving the on screen throttle 1% sufficient movement for a locomotive running in 28 or 14 speed step mode to need to start moving? I don't personally think it should, but your changes would force that.

I understand your point, since I made the same point in February 2021 :) but was voted down.

Just let me clarify: This behavior was in the code for AbstractThrottle, LocoNetThrottle and Pr2Throttle before. I "only" changed the others to match it. That's PR #12423 and it does not fix this issue - I just happened to stumble across that inconsistency while checking those throttles if they're also affected by this issue.

> Ideally, you would make whatever fix you need to only in the WiThrottle code, which may mean you need to make it aware of the current speed step setting of the underlying throttle.

The fix to this issue is in PR #12422 and does not change the speed step 0->1 thresholds.

Heiko

PR #12422 (2023-09-20): Fix #12337
I believe this is the minimum to fix #12337