JMRI/JMRI #12346: java error new install on desktop windows 10 64 bit

davidparks9, 2023-09-03:
[image]
[image]
see photos of java version and this was latest production release install 
just received java error after attempting to run decoder pro or panel pro

danielb987, 2023-09-03:
Which Java version have you installed? The output of `java -version` tells that you have Java 8 and JMRI 5.x needs Java 11.

See https://jmri.org/java

If you already have installed Java 11, you might need to remove Java 8.

davidparks9, 2023-09-07:
thanks it was the java version

closed 2023-09-07 by danielb987