JMRI/JMRI #12380: paragon 4 Broadway limited locomotive union pacific fef-3

davidparks9, 2023-09-07:
everytime i attempt to program decoder it changes the address to 144 for engine
i have tried reading and programming cv's separately,  same results it changes engine address from 840 to 144 not sure why 
programming on program track with digitrax  210

PUlvestad, 2023-09-07:
Without JMRI, can you change the CV's directly from the DCS and a Digitrax throttle?

davidparks9, 2023-09-07:
Yes I can with a dt500 throttle

https://github.com/JMRI/JMRI/issues/12380#issuecomment-1710329807

bobjacobsen, 2023-11-03:
When you get a chance, could you repeat this with the LocoNet monitor open and attach here what it records?  Thank you.

davidparks9, 2023-11-04:
I will but it will probably be December I’m installing a 300,000 light display currently

https://github.com/JMRI/JMRI/issues/12380#issuecomment-1792291790

closed 2024-04-01 by github-actions[bot]