JMRI/JMRI #12386: DS74 DIGITRAX SCRIPT TO PROGRAM IT

ClarkGreg, 2023-09-07:
This is a JMRI script I created to configure Digitrax DS74 including routes by reading a text file.  It first resets the DS74 then reads the text file and updates the DS74  So changes can be done by simply editing the file and reloading.

It does write messages to system console instructing user to press buttons on the DS74.  Also, you need to update the script with the filename of the text file.

It is well commented.

The input file looks like:
ID 200
OPS CV 11
OPS CV 16
RTS 1 121T 201T 202C
RTS 2 122C 203T 204C
RTS 3 123C 205C 206T 
to indicate the ID, any CVs to be set then upto 8 routes.
(Note: File renamed from .py to ,txt)
 
[PGM DS74 GEN.TXT](https://github.com/JMRI/JMRI/files/12551758/PGM.DS74.GEN.TXT)

bobjacobsen, 2023-09-07:
Thank you! PR #12387 will merge this shortly.

closed 2023-09-13 by bobjacobsen

ClarkGreg, 2023-09-13:
Great,

I am happy to have contributed something for the effort.

 

Thanks for the update,

Clark 

 

Closed #12386 <https://github.com/JMRI/JMRI/issues/12386>  as completed.