JMRI/JMRI #12416: Make SVG graphics usable on Panels

bobjacobsen, 2023-09-15:
From a [JMRIusers suggestion](https://groups.io/g/jmriusers/message/220998), it would be good to make SVG icons usage for panels.  Right now, they don't show up in the catalog.

bobjacobsen, 2023-09-15:
If you add SVG to `java/src//jmri/CatalogTreeManager.java` line 107 (`IMAGE_FILTER`), the SVG files start showing up in the catalog with "invisible icon" displayed.  You can still drag and drop them to an icon specific image.  If you do that and place the resulting icon on the Panel Editor panel, it displays nothing (is blank).  

So it looks like getting the NamedImage to display SVG is a first step.

danielb987, 2023-10-31:
It would be good to implement this.

danielb987, 2023-11-02:
> So it looks like getting the NamedImage to display SVG is a first step.

I have support for SVG images in NamedIcon in this branch:
https://github.com/danielb987/JMRI/tree/svg_image

But rotation and scaling are still pixel based, not vector based. I think this document has the solution for that:
https://core.ac.uk/download/pdf/70407874.pdf

danielb987, 2024-12-16:
I think this issue should stay open even if no one have time to work on it for now.