JMRI/JMRI #12429: OperationsCarRoster.xml not saving when RFID deleted

jerryg2003, 2023-09-20:
JMRI 5.5.4 (and earlier):

Editing an entry in the Cars table to remove an RFID tag (by selecting the blank line at the top of the pulldown list) takes effect in the table (although you have to close and open the table to see it -- as reported in PR #12288).  However, the OperationsCarRoster.xml is NOT updated so the previous RFID tag reappears when JMRI is reloaded.   If the RFID tag is changed (rather than removed), then the OperationsCarRoster.xml file IS updated. 

[The OperationsEngineRoster.xml IS updated correctly in both the change and delete cases.]

DanielBoudreau, 2023-09-24:
I believe PR#12445 will fix the saving "no tag" and updating the tables for cars and engines.

DanielBoudreau, 2023-10-08:
Can you confirm fixed in JMRI 5.5.5?

jerryg2003, 2023-10-08:
@DanielBoudreau Confirmed fixed.  Thanks.

closed 2023-10-08 by jerryg2003