JMRI/JMRI #12440: Stale bot not labelling, should be updated to stale action

bobjacobsen, 2023-09-23:
The [stale bot](https://github.com/probot/stale) has a bug where it's not adding the proper labels to issues and PRs that it closes.

We have it configured to add labels (from .github/stale.yml):
```
# Label to use when an issue has been closed
close-issue-label: "Closed due to inactivity"
close-pr-label: "Closed due to inactivity"
```

These are not working: Closed items are just marked as closed as complete.  

While researching this, I found that stale bot [is deprecated](https://github.com/probot/stale) in favor of the [stale action](https://github.com/actions/stale).  Instead of trying to fix stale bot, it would perhaps be better to update to the stale action.

danielb987, 2023-09-23:
> While researching this, I found that stale bot [is deprecated](https://github.com/probot/stale) in favor of the [stale action](https://github.com/actions/stale). Instead of trying to fix stale bot, it would perhaps be better to update to the stale action.

I agree. I have thought about doing that, but I haven't had time to do it.

bobjacobsen, 2023-11-02:
See PR #12576

bobjacobsen, 2023-12-18:
The new setup is properly applied "closed due to inactivity" tags.  See #12428

closed 2023-12-18 by bobjacobsen