JMRI/JMRI #12448: JMRI LCC Hub IOException

dsand47, 2023-09-25:
When JMRI using the "CAN via GridConnect Network Interface" quits, the JMRI running the LCC Hub Server has an exception. 

Both machines are running JMRI 5.5.4.

Client configuration (macOS 13.6):
[image]

Server status (Raspberry Pi OS 6.1.21):
[image]

Server log:
[session lcc hub.log](https://github.com/JMRI/JMRI/files/12719476/session.lcc.hub.log)

dsand47, 2023-09-25:
Quitting JMRI on the LCC Hub server creates similar errors on the client.

[session.log](https://github.com/JMRI/JMRI/files/12719645/session.log)

bobjacobsen, 2023-11-01:
Is this perhaps just an issue of what's being logged?
```
2023-09-25T15:44:32,581 hub.Hub                               ERROR -  [Thread-25]
java.io.IOException: Connection not terminated normally
	at org.openlcb.hub.Hub$ReaderThread.readByteProtected(Hub.java:274) ~[openlcb.jar:?]
	at org.openlcb.hub.Hub$ReaderThread.loadChars(Hub.java:242) ~[openlcb.jar:?]
	at org.openlcb.hub.Hub$ReaderThread.run(Hub.java:206) [openlcb.jar:?]
```
The `java.io.IOException` is routine when a connection is broken from the remote end.  Perhaps this should just be changed to a ERROR/WARN/INFO/DEBUG level (?) message without the traceback.